adafruit/Adafruit_nRF52_Bootloader #7: Use adafruit VID & PID

ladyada, 2018-06-13:
heya @hathach we can make a windows driver file for the nrf52840 but please use our VID/PID
right now it seems to be set to 0xCAFE & 0x4011

[image]

Please use Feather nRF52840  VID 0x239A    PID 0x0029   # UF2 bootloader (CDC+MSC)
thank you!

hathach, 2018-06-14:
Yeah, @microbuilder  already told me the VID/PID, though, since the bootloader may still change 
 the usb interface. It is better to do this last, otherwise it could mess up with windows driver causing instant reset if the incorrect interface with a known VID/PID is plugged.

hathach, 2018-06-14:
Meanwhile we could use this temporary driver inf for windows here for testing. Though I will try to finialize the interface asap.
[WinCDCdriver.inf.zip](https://github.com/adafruit/Adafruit_nRF52840_Bootloader/files/2101769/WinCDCdriver.inf.zip)

hathach, 2018-06-15:
Ah thanks, I could access it now. I am finalizing the usb interface. I think we should have an extra custom interface although we didn't do much with it now, but it will allow us to have option to send command or communicate with feather without changing pid later. (Dual com port is too confusing as wiced has proved). Will update asap

ladyada, 2018-06-15:
Yes I agree! i think for circuitpython single CDC UART for REPL has worked very well on M0 and M4 boards. It's true it is not great for raw data transfer but not many have complained :)
for M0 we have a lot of endpoints, so we have CDC UART + Mass Storage for disk access + HID for keyboard/mouse emulation. That seems to work for everyone
how many endpoints are there on nrf52840?

hathach, 2018-06-15:
Nrf52804 has max 8 endpoints in & out.
- Cdc use 2 in, 1 out. 
- Msc use 1 in, 1 out
- one of custom will takr 1 in, 1 out

There is still 4 more pairs left, 1 of those can be used for HID in user Arduino sketch. There is plenty left :)

ladyada, 2018-06-15:
ok! i think lets start with just CDC uart, then look to add MSC. Finally we can try HID or other UART later in the future. please let me know if you need help testing or debugging! 
we have a few DevKits and got the bootloader loaded easily and can communicate over JLink serial for debugging work :)

hathach, 2018-06-15:
Usb and msc already work rather reliably, i will push out update that finalize the pid. You could update the driver, signed etc.. So that we could test it easier on windows. 

Left work for usb would be uf2 integration and make sure it work reliably across OS.

microbuilder, 2018-06-15:
@hathach Just a heads up that Limor has some MSC issues on Windows 7 that are worth looking into. Transfers would fail once or twice before working the third time, etc. Probably just something really minor, but worth trying to reproduce or getting a USB capture from Limor.

ladyada, 2018-06-15:
yep! will do once i get a more comfy setup - ill open up issues with debugging details!

hathach, 2018-06-15:
That is strange, I tested with windows 10 and it seems to be fine, I will pull out an windows 7 machine to test.

hathach, 2018-06-16:
I have updated the VID/PID to match the list, the usb is composed of
- cdc 
- msc
- custom class ( for whatever purpose we decide to use later, maybe a command channel to factory reset/clear filesystem etc ..) 

I think it is decent enough for bootloader

@ladyada please make a windows driver file, that will help other to easily test out the bootloader. Bootloader is update in Arduino repo already, I will update the circuitpython in my folk and submit an PR later along with UF2 implemenation.

ladyada, 2018-06-18:
sure, can you commit your latest version to the adafruit/circuitpython usbboot branch, or give me a hex to load onto this DK, so i can test the windows driver?

hathach, 2018-06-19:
I just pushed the update bootloader binary to nrf52840 usbboot branch. It should fix your previous issue with failed uploading as well. To flash bootloader please type

  make V=1 BOARD=feather52840 bootloader

to upload with usb cdc

  make V=1 BOARD=feather52840 SD=s140 all
  make V=1 BOARD=feather52840 SD=s140 SERIAL=/dev/ttyACM1 dfu-gen dfu-flash

The cp port for feather52840 has some issue with uart driver ( i guess), you should see the REPL prompt but couldn't type anything yet. We will fix that later after the uf2 implementation :D

ladyada, 2018-06-25:
ok by the way i made the UF2 bootloader serial .inf driver

https://github.com/adafruit/Adafruit_Windows_Drivers/commit/25f6d8d8dc5aa838d53841ed4a281b9be1510663

i can change it eaily and then sign it for windows 7. for windows 10/mac/linux no driver is needed :)

hathach, 2018-06-25:
That will make thing easier for everyone :) btw, could we map the custom interface ( reserve now ) for libusbk with our signed driver. Currently, I use Zadig for windows binding, Mac/Linux doesn't need that.

[image]

ladyada, 2018-06-25:
good question, i will research it - what do you want the custom interface for, is it debug output like we had for WICED?
FYI - for samd21/51 we use 2 endpoinds for HID keyboard/mouse.

hathach, 2018-06-25:
I am not sure for now, it depends on later need. It can be used as extra comm channel. Maybe issue an factory reset, read memory contents etc... But that custom interface does not need to be part of our signed driver at all. It presents only so that you don't have to make another VID/PID later on.

nrf52840 has 7 pairs of bulk/int endpoints. So far only 4 In, 3 out are used.It is still  Btw, all HID can be merged to use a single In & out endpoint. 

PS: If you don't really feel the need to include the custom interface, we could remove it. We could always add it later if needed, though you need to make another pair of VID/PID since the usb interfaces are changed.

ladyada, 2018-06-25:
the custom interface is good for now - especially if it helps you with debugging :) 
long term we can think about how we want to do debugging on complicated devices like nrf52 with a radio stack - there is no obvious way to do it so i will think about it :)

dhalbert, 2018-06-25:
@hathach:
>Note: The REPL prompt is printed but It won't interact, I will fix that later, possibly uart driver issue.
I am seeing something like this problem on Linux but not on Windows or Mac, when I use the JLInk serial support. If I connect via a terminal program, I can't type at the board. If I press reset, I see part of the startup prompts, but still can't use it. I think there might be some kind of bug in the jlink support for Linux (or a Linux USB CDC bug of some kind). But the JLinkGDBServer seems to work.

ladyada, 2018-06-25:
@dhalbert the CDC half uses usbser, so that part works on win10 right now. the special interface needs libusb on win7 and maybe win10 because its not a standard windows driver. whee :)

hathach, 2018-06-25:
@ladyada for debugging we should bring out the jlink and use the segger RT for tracing event, I haven't used much but it looks useful. Maybe when I got more familliar with cp build management, I will try to make a segger studio project for nrf52 to help with debugging :) .

I am thinking the custom interface is for Adafruit-specific command such as an AT variant, to retrieve board information, command nrf e.g send a specific payload to a specific address etc...but we may not need it at all :(.

hathach, 2018-06-25:
@dhalbert Ah thank you clarification, I am indeed linux user, I did wonder why I haven't got any complains on this from ktwon lol :)) . I did try to troubleshoot it but couldn't find a clue. But we will switch to the usb cdc on cp anyway, maybe it is not worth the effort.

Like Limor says, libusbk is only for the vendor specific (custom) interface. Registering it with libusbk with windows allow us to write a simple c/python code to communicate with it directly as separate channel (from cdc). Dual cdc port is very frustrating as wiced shows :( since the port number is quite random.

ladyada, 2018-06-25:
another thing we can discuss much later is webusb - it is HID-like (and needs libusbk or similar) but then you can connect over a webbrowswer. just thinking outloud, no need to implement right now :)

hathach, 2018-06-26:
@dhalbert thanks a lot for the link, though since we are moving to the usb cdc of the chip without using the jlink cdc interface, I think I will just skip that for now. We can come back to this later on.

@ladyada  oh, webusb is new to me, is it documented here https://wicg.github.io/webusb/ . I will check its requirements to see if we need to prepare anything for it.

ladyada, 2018-06-26:
ok sounds good! don't worry about implementing webusb yet - its just normal 2-endpoint USB but with a special descriptor that Chrome browser can find and communicate with.

hathach, 2018-07-12:
@ladyada please also add the entry for usber in circuitpython vid/pid as well

PID 0x802A   # CircuitPython application mode (CDC+MSC)

ladyada, 2018-07-12:
hooray! @dhalbert can you add and re-spin/re-release?

hathach, 2018-07-12:
Let's do it for the Arduino application as well :D 

summary from the list
Feather nRF52840  VID 0x239A    
                                PID 0x0029   # UF2 bootloader (CDC+MSC)
                                PID 0x8029   # Arduino application mode (CDC+MSC)
                                PID 0x802A   # CircuitPython application mode (CDC+MSC)

ladyada, 2018-07-12:
OK we will do it!

dhalbert, 2018-07-12:
OK, I'll have a PR on master shortly.

dhalbert, 2018-07-12:
I'm sorry - I was confused. You want this on this one the Windows Drivers package, right?

dhalbert, 2018-07-12:
I actually already added this in the latest Windows Drivers package. Download this one:
https://github.com/adafruit/Adafruit_Windows_Drivers/releases/download/2.3.0/adafruit_drivers_2.3.0.0.exe

```
"%FEATHER_NRF52840% UF2 Bootloader (0029:00)"=DriverInstall, USB\VID_239A&PID_0029&MI_00
"%FEATHER_NRF52840% UF2 WebUSB dummy (0029:04)"=NullInstall, USB\VID_239A&PID_0029&MI_04
"%FEATHER_NRF52840% Arduino (8029:00)"=DriverInstall, USB\VID_239A&PID_8029&MI_00
"%FEATHER_NRF52840% CircuitPython (802A:00)"=DriverInstall, USB\VID_239A&PID_802A&MI_00
```

dhalbert, 2018-07-12:
Fixed by https://github.com/adafruit/Adafruit_Windows_Drivers/commit/47196108972a4a907c50eeca7cc065c57abf04e3

hathach, 2018-07-12:
@dhalbert  is there any problem if the webUSB dummy is not implemented !! Maybe we should add it later, anyway, I will check out to see what is webusb tomorrow.

dhalbert, 2018-07-12:
No problem! The "null" dummy device driver for WebUSB is only used if it shows up -- it's not required. Don't spend a lot of time on webusb right now: we experimented with it but there are various problems to solve first.

hathach, 2018-07-12:
Ok, I got it

hathach, 2018-08-15:
@ladyada Do you think we should have an additional CDC only for bootloader (without MSC UF2) when using with touch 1200 to enter DFU. Since MSC takes longer to enumerate (~2 seconds), and it is kind of annoying when File Explorer pop up for a few seconds everytime Arduino upload a sketch. Some OS may complain with Safe Remove message as well.

ladyada, 2018-08-15:
no, i think we should stick to plain UF2. we use it a lot with all our SAMD21 boards and we've gotten no complaints!

hathach, 2018-08-15:
@ladyada sorry, I didn't express it well enough. UF2 still show up when we are holding the DFU button or in any other ways we are using. I am talking about an addition mode where the bootloader will show up only as Serial for Arduino usage. Basically, it is done by writing to NRF_POWER->GPREGRET and soft-reset when touch 1200 happens.   
https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/cores/nRF5/wiring.c#L65

To sum up
- CDC + MSC when holding DFU button or softreset with UF2_MAGIC
- CDC only when softreset with SERIAL_MAGIC and touch 1200
- OTA: when softreset with OTA_MAGIC and or triggered by BLE event

ladyada, 2018-08-15:
lets stick to only two modes:

    CDC + MSC when holding DFU button or softreset with UF2_MAGIC
    OTA: when softreset with OTA_MAGIC and or triggered by BLE event

we can adjust later if we get complaints but so far nobody complained about M0 and we use the same bootloader with CDC+MSC on both

hathach, 2018-08-15:
Sigh, the CDC-only is only entered when we dictate it. Default mode is CDC + UF2, I really don't like to wait an extra 5 seconds with touch 1200 whenever uploading a sketch ~.~

ladyada, 2018-08-15:
i dont see why it needs 5 seconds tho? it doesnt with m0 :/ only the first time takes a little time. afterwards its instant

hathach, 2018-08-15:
I am not enitrely sure, I wait for 1.5 seconds now but it fails occasionally especially when uf2 drive doesn't show up yet while starting uploading. I guess while host is reading the msc uf2 contents (the firmware fiile) that could cause some sort of delay in response when we do dfu via serial. It could be usb stack's bug though. Since we uploading with serial it could be beneficial to just show up as cdc only. This is done by simply supply the host with cdc-only descriptor instead of composite descriptor. All the code remains the same.

dhalbert, 2018-08-15:
This delay does not show up on the Atmel USB stacks, so it's probably worth looking at some USB traces, including timing, to see what the delay is. I thought also some people are seeing a significant delay in CPy on nRF52 before the REPL connection is available, and that would be TinyUSB also, I think, since we don't see that delay in Atmel CPy either.

I have been very happy using Wireshark with USBpcap both on Linux and Windows to debug this kind of thing. Beagle is OK but does not parse the non-descriptor packets without paying more money.

hathach, 2018-08-15:
@dhalbert the first connection with serial is delayed only for CP, I run cdc with several other projects including the bootloader but doesn't encounter that issue. I probably didn't get it integrated well with CP yet. It is a different issue though.But since it is purely application code, we could always fix it later. 

For the tracer, I got a really good hardware analyzer given by KTown. Under the analyzer we only see the device NAK the IN token from host, the usb stack is not perfect I admit. The nrf5x port is only run as proof of concept just a few weeks ago, the pre-production IC is full of bug that making the sw work is impossible.

hathach, 2018-08-15:
I just pushed the touch option for adafruit-nrfutil
https://github.com/adafruit/Adafruit_nRF52_nrfutil/commit/18fa21a36c682953267515bc94991ca50882d992#diff-eeb4dae138b55a43848ee839b4e10a96R107

Our scenario is adafruit-util wiill open and disconnect circuitpython on nrf52840 at 1200 bps, cp will reset to DFU mode.  `TOUCH_RESET_WAIT_TIME` is time for util to wait for nrf52 is enumerated as bootloader. Currently it is 1.5 second, but not enough. Often than not I see the dfu process start before the file explorer could show up the thumbdrive. Sometimes the dfu process would failed.What is my mind is
1. increase time delay 
2. reduce delay time by report to host as cdc only, msc take lots more time to enumerate
3. try to hunt down the possible cause, which could take a bit of time :(

ladyada, 2018-08-15:
ok i understand.  thank you for explaining it to me, thach!

i think we should eventually try to figure this out to reduce but for now we can have the CDC only option with 1200 baud touch. hopefully we'll be able to fix it later :)

hathach, 2018-08-15:
@ladyada Yeah it is confusing, and I didn't express it very well. I admit I have issue with expression due to the remote work and lack of communication with human :( .

I will also need your help to have an extra VID/PID for the bootloader with CDC only. Like you said, later on, if we could hunt down the issue and the delay time is under 2 seconds, we could remove that mode. User won't ever notice it. 

PS: Thinking about it, it is more and more the usb stack issue since we having lot of data transfer in both interfaces in a really short time. Some conflict must occur. I will try to trace it, but we really need a working thing with Arduino.  As mentioned before, Arduino repo use freeRTOS, I haven't tested usb stack with freeeRTOS for ages, and the last time I test is with a very old version :(.

hathach, 2018-08-15:
PS: we have another pending issue  with bootloader in OTA as well. The same code doesn't run with SD 6.1.0 while it runs with 6.0.0 despite release note says they are binary compatible 
https://devzone.nordicsemi.com/f/nordic-q-a/37473/advertising-does-not-start-with-s140-6-1-0-same-hex-run-without-issue-with-s140-6-0-0

Have been tracing it in the last couple of days, at worst we will release it with 6.0.0 first.

hathach, 2018-08-15:
@ladyada is it OK that I use the `PID 0x002A` for the bootloader with CDC only. Any number will do though.

ladyada, 2018-08-15:
hiya, im back -  yep 0x002A is avilabale. lets try this! :)

```
Feather nRF52840  VID 0x239A    PID 0x0029   # UF2 bootloader (CDC+MSC)
                                PID 0x002A   # Arduino bootloader (CDC)
                                PID 0x8029   # Arduino application mode (CDC+MSC)
                                PID 0x802A   # CircuitPython application mode (CDC+MSC)
```

hathach, 2018-08-15:
Thanks, I will update it asap

ladyada, 2018-08-15:
awesome, you're doing great :)

hathach, 2018-08-15:
Thanks, I am hoping to nail it down for other to test. Bootloader is the only thing we need to get it right in the first time :)

hathach, 2018-08-16:
changing to CDC only mode, I could get 10 out of 10 consecutive uploading. 1.5 second is probably long enough for linux. I will test with mac and windows as well. Please update the windows driver exe whenever you feel convenient :) .I am cleaning up and pushing it out :D

ladyada, 2018-08-16:
ok sounds good!

hathach, 2018-08-24:
@ladyada  can you update the driver exe with the new PID for Arduino bootloader, I need it to test with windows uploading