adafruit/Adafruit_nRF52_Bootloader #8: Automate builds

ladyada, 2018-06-23:
Hey Thach, since you're doing a ton of work in this repo would it be helpful to have automatic builds on commits via travis?

Dan & I could get that going for you so that the hex files appear on S3 - then you wouldn't have to keep committing binaries as you build and we would have less risk of out-of-sync bootloaders between this and circuitpython repo!

Please let us know :)

hathach, 2018-06-23:
Got it, no problem, I will add the .travis file to auto build. Then ping you for the S3 upload script.

Note: bootloader + SD combo is upgradable, we should keep an binary for each "release verion", so far nrf52832 have v2 and v5 (there is v6 coming to be consistent with nrf52840).

dhalbert, 2018-07-03:
@hathach Just to be clear you haven't started on travis-ing this yet, have you? I will work on this but just want to make sure I'm not duplicating something you haven't checked in yet.

hathach, 2019-01-31:
I guess it is safe to close this now. Feel free to open if it isn't

closed 2019-01-31 by hathach