adafruit/Adafruit_nRF52_Bootloader #24: UF2 to upgrade bootloader + SD itself

hathach, 2018-08-23:
So far only DFU Serial could upgrade Bootloader + SD combo, uf2 should also do this as well. With a set of specific requirement e.g
- family option is different than one used by application firmware
- more ...

dhalbert, 2018-08-27:
On atmel-samd UF2, the UF2 bootloader cannot update itself directly. Instead, you load a update program into non-bootload (user) space which then rewrites the bootloader. See the update-bootloader*.ino and .uf2 files in the UF2 release: https://github.com/adafruit/uf2-samdx1/releases/latest

hathach, 2018-08-27:
it is doable with nrf52, we are doing it with cdc interface. it could upgrade itself via cdc packet. Maybe it is not too difficult to get it work with uf2

hathach, 2019-11-28:
bootloader + SD uf2 image will have an different uf2 family ID to prevent accident or corrupted image from normal application uf2.

closed 2020-05-29 by hathach