adafruit/Adafruit_nRF52_Bootloader #26: Double-click reset possible?

dhalbert, 2018-08-29:
@hathach @ladyada @tannewt 

We were talking a few days ago about not being able to detect double-click in the bootloader because the nRF52 clears out everything on a hard reset. However, the [nrf52840 data sheet](http://infocenter.nordicsemi.com/pdf/nRF52840_PS_v1.0.pdf) in section 5.3.6.8 (page 69) says:

>Note: The RAM is never reset, but depending on a reset source the content of RAM may be
corrupted.

So I just want to double-check if someone has actually tried this and confirmed that leaving a sentinel value in a RAM location doesn't work.

hathach, 2018-08-29:
Last time I tried, it is totally failed, i will give it another with multiple sram bank as scratch data

hathach, 2018-08-30:
@dhalbert @ladyada  I just get the double reset work with nrf52. Yeah, you guys are so right suspecting my failed attemppt previously kk. I think the problem I run into previously is first trying with some register, then try to use a random SRAM. but the SRAM is clear by startup script ( not hardware). We need to specify the linker with NOLOAD to spare that SRAM location. Feel so good now. I will clean up and push the update.

hathach, 2018-08-30:
There is lots of works for releasing nrf52832 feather back then, so hahaha, I gave up so soon on this. Thanks for bringing this up :D

hathach, 2018-08-30:
Update:  the double reset really **DOES NOT** work with nrf52832, only work with nrf52840, same code. No matter how I tried :D  There is a reason for my giveup  afterall  !!!

ladyada, 2018-08-30:
wow! ok well we have auto-bootloader for '532 so lets add for nrf52840 only

dhalbert, 2018-08-30:
that's weird, hope it continues to work! Thanks for the thorough testing. We should try it on multiple 52840 boards to make sure.

hathach, 2018-08-31:
Yeah, please test it to see if that work.  840 work with my first try, so I think it will be OK.

dhalbert, 2018-08-31:
Works! Great!

hathach, 2018-08-31:
Yeah, the 840 works right away. Currently 832 share same code (I didnt #ifdef to opt it out). Still it refuses to work. I think I did put 832 under debugger to test sram before, and convinced that  its sram got clear. I will check noridc devzone, I remembered seeing a relevant topic back then.

dhalbert, 2018-08-31:
Works on my two PCA10056 boards. I think we can declare success!

closed 2018-08-31 by dhalbert

ladyada, 2018-09-01:
tested on my PCA - works great :)

ladyada, 2018-09-01:
tested on Feather 840 as well - works great for doubleclicking
i updated the guide as well https://learn.adafruit.com/circuitpython-on-the-nrf52