adafruit/Adafruit_nRF52_Bootloader #29: Generate separate "update bootloader" image

dhalbert, 2018-09-07:
Right now the bootloader can self-update, but if there is no existing UF2 bootloader, or it's corrupted, you have to use the SWD interface to initially load the bootloader.

Generate an "update-bootloader" image that can be loaded with Nordic's bootloader or this bootloader, which will replace the existing bootloader.

https://github.com/adafruit/uf2-samdx1 provides this capability with the `update-bootloader-*.{bin.uf2,ino}` files it generates. See `selfmain.c` and the `SELF` targets in the `Makefile` for how it does this. The `.ino` is an Arduino sketch that does the same thing, so you can update the bootloader even if all you have is just the Arduino IDE and you don't know how to use the nrf command-line tools to load an image. That would be an optional feature for this issue.

Issue filed at @hathach 's request.

hathach, 2018-09-07:
@dhalbert in case the bootloader is corrupted, we must use a jlink to debrick it :D since there is no way to load this app to rewrite the bootloader :D . This would be mainly use to replace whatever bootloader on the board (that could load this image) by our bootloader.

hathach, 2018-09-07:
opps, I just realized you mean to run an in-place binary script in filesystem to update the bootloader ?

dhalbert, 2018-09-07:
No, not in filesystem: we'd use the existing bootloader to load a program that rewrites the bootloader. This program would overwrite CircuitPython or an Arduino sketch or whatever was in flash. It would be supplied .uf2 and .hex format (and maybe .ino).

hathach, 2018-09-07:
Ah ok, I was thinking on how to correct the current corrupted bootloader. But I guess let use jlink, this application is only to replace whatever an oem bootloader on a nrf52840 board to our bootloader.

hathach, 2018-09-07:
This is a bit tricky and easy to brick, we will only create one version of this app for a specific board with current sd 6.1.0 as current. User needs to go through normal dfu-flash afterward to upgrade bootloader to the latest. Hopefully nordic bootloadrr does not use locked, or any kind of memory protection :D

dhalbert, 2018-12-06:
@hathach Does the bootloader protect itself from a user-program overwrite by using the ACL peripheral? I see that once you protect a region by writing the ACL registers, they can't be changed back until there's a hard reset.

I thought perhaps this is how the bootloader protects itself, and if we can get the bootloader to skip the ACL-setting step (e.g. via the second switch or a pin check or something, then we could enable a user-mode bootloader update.

hathach, 2018-12-06:
@dhalbert currently ACL is not enabled yet, There is an issue to enable ACL here. https://github.com/adafruit/Adafruit_nRF52_Bootloader/issues/21 . So yeah, we are free to do user-mode bootloader update.

jerryneedell, 2019-09-26:
Jus to clarify, Can a user now update the Bootloader by simply creating a .uf2 from the released .hex file and then dragging it to the device? If so, should the release assets include .uf2 versions?

hathach, 2019-09-26:
@jerryneedell current uf2 cannot be used to upgrade the bootloader, bootloader can only be upgraded using cdc interface. Best to use Arduino IDE to do so. 

But we could add this feature in future release. Just making sure it doesn’t brick device 😁😁

hathach, 2019-09-26:
you can also use the adafruit-nrfutil manually  as well, i will update this comment with the correct syntax when I got home.

jerryneedell, 2019-09-26:
@hathach  Thank you! That would be very helpful!

hathach, 2019-09-26:
> @hathach Thank you! That would be very helpful!

here it is
`adafruit-nrfutil dfu serial --package bootloader.zip -p /dev/ttyACM0 -b 115200 --singlebank --touch 1200`

jerryneedell, 2019-09-26:
@hathach  Thanks -- before I try it, can you please confirm that I can do this with a CircuitPlayground Bluefruit connected to my system via a USB cable. Do I have to manually enter bootloader mode or any other pre-configuration?

hathach, 2019-09-26:
> @hathach Thanks -- before I try it, can you please confirm that I can do this with a CircuitPlayground Bluefruit connected to my system via a USB cable. Do I have to manually enter bootloader mode or any other pre-configuration?

You don't need to force it into DFU mode, the command will does the that trick by connecting at baud 1200 bps then disconnect `--touch 1200`. It will then could upgrade the bootloader as normal. You can add --verbose option  right after adafruit-nrfutil as well. Don't reset/disconnect or do anything until you see  **Device programmed**

https://learn.adafruit.com/bluefruit-nrf52-feather-learning-guide/updating-the-bootloader

jerryneedell, 2019-09-26:
worked great! both on CircuitPlayground Bluefruit and feather_nrf52840_express

hathach, 2020-05-29:
bootloader can be updated using uf2 and cdc now. For non-adafruit such as Nordic bootloader and other vendor bootloader, that will mostly use ACL to protect the bootloader. I feel this issue is out of date, could be reopened or create new one if needed.

closed 2020-05-29 by hathach