adafruit/Adafruit_nRF52_Bootloader #32: Compiling error

seihan, 2018-10-22:
By trying to build the bootloader for nrf52832 I've got following error message:

make BOARD=feather_nrf52832 all
`CC main.c
In file included from lib/softdevice/s132_nrf52_6.1.0/s132_nrf52_6.1.0_API/include/nrf_sdm.h:55:0,
                 from lib/sdk11/components/libraries/bootloader_dfu/dfu_types.h:27,
                 from lib/sdk11/components/libraries/bootloader_dfu/bootloader.h:27,
                 from src/main.c:43:
lib/softdevice/s132_nrf52_6.1.0/s132_nrf52_6.1.0_API/include/nrf_soc.h: In function 'sd_evt_get':
lib/softdevice/s132_nrf52_6.1.0/s132_nrf52_6.1.0_API/include/nrf_soc.h:807:1: error: stack usage computation not supported for this target [-Werror]
 SVCALL(SD_EVT_GET, uint32_t, sd_evt_get(uint32_t * p_evt_id));
 ^
cc1: all warnings being treated as errors
make: *** [Makefile:448: _build-feather_nrf52832/main.o] Fehler 1
`

I've tried with gcc-arm-none-eabi 5_2-2015q4 and 7-2018-q2-update

jerryneedell, 2018-10-22:
It builds for me on an Ubuntu 18.04 system with the tools from (GNU Tools for Arm Embedded Processors 7-2018-q3-update)  -- you may want to try upgrading your toolset.

snue, 2018-10-22:
Also works on Fedora with the distribution provided arm toolchain:

```
$ dnf install arm-none-eabi-gcc-cs-c++ arm-none-eabi-newlib
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (Fedora 7.1.0-5.fc27) 7.1.0
```

hathach, 2018-10-23:
@seihan could you state which OS you are on ?

seihan, 2018-10-23:
Thanks a lot!
It also works for with the distribution provided arm toolchain.

`arm-none-eabi-gcc (Fedora 7.1.0-5.fc27) 7.1.0
`

closed 2018-10-23 by seihan

verjus, 2020-06-07:
Had the same problem with a recent version of compiler and had to remove "-fstack-usage" from Makefile.