adafruit/Adafruit_nRF52_Bootloader #35: DFU OTA always fails (nRF52840 board)

beegee-tokyo, 2018-12-12:
Someone has an idea? Am I using the tools wrongly?

DFU OTA update over nRFutil always fails after
```
[DFU] Response received (Op Code = 2, Status = 6)
[DFU] Remote DFU error: OPERATION FAILED
```

Build environment:
Hex file is created with Arduino IDE V1.8.7
Adafruit nRF52 board version 1.0.2
Board has Bootloader/Softdevice **`pca10056_bootloader_s140_6.1.1r0`**
(Tried feather_nrf52840_express variant, but no difference)

Creation of DFU OTA package:
**`adafruit-nrfutil dfu genpkg --application PPG-Prod.ino.hex  Raytac-Prod-52840.zip`**
Content of created manifest.json:
``` JSON
{
    "manifest": {
        "application": {
            "bin_file": "PPG-Prod.ino.bin",
            "dat_file": "PPG-Prod.ino.dat",
            "init_packet_data": {
                "application_version": 4294967295,
                "device_revision": 65535,
                "device_type": 65535,
                "firmware_crc16": 11098,
                "softdevice_req": [
                    65534
                ]
            }
        },
        "dfu_version": 0.5
    }
}
```

nRF52 log:
``` LOG
nRF Connect, 2018-12-12
PPG-1A0C7447CE724A66 (F3:5E:96:95:DF:3B)
V	19:37:45.216	Connecting to F3:5E:96:95:DF:3B...
D	19:37:45.216	gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferred PHY = LE 1M)
D	19:37:45.466	[Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
D	19:37:45.467	[Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	19:37:45.467	Connected to F3:5E:96:95:DF:3B
V	19:37:45.494	Discovering services...
D	19:37:45.510	gatt.discoverServices()
I	19:37:45.931	Connection parameters updated (interval: 30.0ms, latency: 0, timeout: 2000ms)
I	19:37:46.233	Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
D	19:37:46.711	[Callback] Services discovered with status: 0
I	19:37:46.711	Services discovered
V	19:37:46.777	Generic Access (0x1800)
- Device Name [R W] (0x2A00)
- Appearance [R] (0x2A01)
- Peripheral Preferred Connection Parameters [R] (0x2A04)
- Central Address Resolution [R] (0x2AA6)
Generic Attribute (0x1801)
- Service Changed [I] (0x2A05)
Client Characteristic Configuration (0x2902)
Device Firmware Update Service (00001530-1212-efde-1523-785feabcd123)
- DFU Packet [WNR] (00001532-1212-efde-1523-785feabcd123)
- DFU Control Point [N W] (00001531-1212-efde-1523-785feabcd123)
Client Characteristic Configuration (0x2902)
- DFU Version [R] (00001534-1212-efde-1523-785feabcd123)
Device Information (0x180A)
- Model Number String [R] (0x2A24)
- Serial Number String [R] (0x2A25)
- Firmware Revision String [R] (0x2A26)
[... 130 lines cut ...]
I	19:38:12.331	[DFU] Data written to 00001531-1212-efde-1523-785feabcd123, value (0x): 02-01
A	19:38:12.331	[DFU] Initialize DFU Parameters completed
I	19:38:12.331	[DFU] Notification received from 00001531-1212-efde-1523-785feabcd123, value (0x): 10-02-06
A	19:38:12.331	[DFU] Response received (Op Code = 2, Status = 6)
E	19:38:12.347	[DFU] Remote DFU error: OPERATION FAILED
V	19:38:12.347	[DFU] Writing to characteristic 00001531-1212-efde-1523-785feabcd123
D	19:38:12.347	[DFU] gatt.writeCharacteristic(00001531-1212-efde-1523-785feabcd123)
D	19:38:17.403	[Callback] Connection state changed with status: 8 and new state: DISCONNECTED (0)
E	19:38:17.403	Error 8 (0x8): GATT CONN TIMEOUT
I	19:38:17.403	Disconnected
A	19:38:17.408	[DFU] Reset request sent
V	19:38:17.441	[DFU] Disconnecting...
D	19:38:17.442	[DFU] gatt.disconnect()
I	19:38:17.442	[DFU] Disconnected
D	19:38:17.442	[DFU] gatt.refresh() (hidden)
D	19:38:17.442	[DFU] gatt.close()
D	19:38:17.442	[DFU] wait(600)
D	19:38:17.472	[Broadcast] Action received: android.bluetooth.device.action.ACL_DISCONNECTED
D	19:38:17.489	[DFU] [Broadcast] Action received: android.bluetooth.device.action.ACL_DISCONNECTED
D	19:38:18.047	gatt.close()
D	19:38:18.050	wait(200)
V	19:38:18.252	Connecting to F3:5E:96:95:DF:3B...
D	19:38:18.252	gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferred PHY = LE 1M)

```

hathach, 2018-12-12:
the package zip file is also created by Arduino (next to hex file), can you use nRF connect/toolbox app to perform OTA with that file.

beegee-tokyo, 2018-12-13:
I tried the .zip file. Different error message, but still no success. Now it says:
``` LOG
[DFU] Receive Firmware Image request sent
[DFU] Uploading firmware...
[DFU] Sending firmware to characteristic 00001532-1212-efde-1523-785feabcd123...
[DFU] Notification received from 00001531-1212-efde-1523-785feabcd123, value (0x): 10-03-06
[DFU] Response received (Op Code = 3, Status = 6)
[DFU] Remote DFU error: OPERATION FAILED
```
The file itself is ok. I can flash it over USB or convert to uf2 and flash it by copying it to the 52840's drive after plugin to PC.

I tried as well to force the bootloader into DFU OTA (pins 15 & 19 to GND) so it shows up as AdaDFU. But still same error.

==========================

It's not the bootloader!
I first tried to update from my ne Huawei M5 tablet, Android 8.0.0. There I get the above error.
Then I tried from my old Samsung Galaxy Tab 10.1, Android 7.1.2. And it works without problems.

It has something to do with either the Huawei tablet or Android 8.0.0. Neither nRFConnect nor nRF Toolbox can do OTA update on the Huawei tablet. The Samsung tablet has no problems with either.

Looking on the error message I get in nRF Toolbox, it seems to be a timeout in the communication. On the Huawei tablet it start uploading (1%) then stops and reports error.

==========================

Closed as it is not a bootloader problem.

closed 2018-12-13 by beegee-tokyo

jewcbox, 2019-08-06:
This is an issue with the bootloader. 

Download this source code and modify 

The resolution was to modify src/sdk_config.h and increase the value of HCI_RX_BUF_QUEUE_SIZE to 16 resolved all issues.

For the developers, I have a handful of Adafruit nrf52840 boards. I traced hci_mem_pool_rx_produce() throwing an error NRF_ERROR_NO_MEM due to m_rx_buffer_queue.free_window_count == 0
In that trace, the value of `length` was 14.

beegee-tokyo, 2019-08-06:
I got it to work by changing the Nordic DFU library and build my own Android app to do OTA DFU without problems with the original bootloader.

Will try your changes as well. Working on a modified bootloader based on this code for my custom nRF52832 modules.

Thanks for pointing to this.

happy49sky, 2019-09-10:
Hi, @beegee-tokyo  
Could you please share your change ?

beegee-tokyo, 2019-09-11:
The only change I made in the Nordic DFU library is in DfuBaseService.java. I removed two fixed delays in the code and instead removed bonding of the device. OTA DFU always failed if the nRF52 was bonded with the Android device.
Here is the DIFF:
[image]

hathach, 2019-09-11:
thanks @beegee-tokyo we will check to see what we could do about this.

beegee-tokyo, 2019-09-11:
@hathach    
As said before, I don't think it is a bootloader problem. It works perfect on Android 5. Only Android 8 and 9 showed the problem.

happy49sky, 2019-09-12:
@beegee-tokyo   Thanks !

fanoush, 2019-09-20:
Hello, I've seen similar/same issue on  52832 based board, I tried to recompile bootloader with HCI queue size set to 16 however there is not enough RAM on 52832 for this (linker fails with ram overflow into stack). What seems to be related and helped me is to decrease number of packets sent by nrfConnect for Android. There is Settings menu with DFU options and there I lowered number of packets from 10 to 5. Looks like 10 packets is too much for queue size of 8?

evandavey, 2021-07-05:
Confirming that I required this change (HCI_RX_BUF_QUEUE_SIZE = 16 in src/sdk_config) for OTA DFU to work using nRF Connect on Android using a Sparkfun MicroMod nRF52840.