adafruit/Adafruit_nRF52_Bootloader #42: Any way to secure against OTA DFU?

lbdroid, 2019-01-17:
On instructions https://github.com/adafruit/Adafruit_nRF52_Arduino/issues/218#issuecomment-455039283, I'm re-posting this issue against what is apparently a more appropriate repository.

Looking through the literature on these as carefully as I can, I've come across reference to a mode referred to as "OTA DFU", which can apparently be triggered by Nordic's nRF Toolbox application, allowing the storage memory to be rewritten over an unsecured bluetooth connection.

I've also seen indication that there is absolutely no security on this functionality at all: #162 (comment)

What I need to be able to do is disable this mode altogether, or in the very least, be able to program it with an update key that I (and only I) am in control of. This "feature" otherwise presents a security hole that makes it completely unsuitable for any application that has even fringe security related implications.

rabih9780, 2019-07-10:
Do you guys have an update on this?

ladyada, 2019-07-10:
no ETA

rabih9780, 2019-07-10:
What would it take to implement secure DFU into this library? upgrade sdk?

ladyada, 2019-07-10:
a complete rewrite

lbdroid, 2019-07-10:
According to this comment; https://github.com/adafruit/Adafruit_nRF52_Arduino/issues/218#issuecomment-479990008

The issue of somebody randomly taking over your device via unsecured DFU mode has at least been mitigated. DFU mode is not enabled unless you explicitly enable it.

In other words, while DFU mode is still not implemented securely, at least you have to tell it to switch to that mode before it is possible to rewrite the firmware using it.

So if you need to implement DFU updates, you could arrange it so that you need to send it a secure command to switch DFU mode on, then use DFU mode, and reboot to resume DFU-disabled mode.

tusharvb19, 2020-09-17:
Hi, any update on secure DFU?

CSC-Sendance, 2023-03-28:
Hi, sorry for necromancing this. 

We have been investigating this issue increasingly and found that WebBluetooth explicitly blocks Nordic's update service, making it impossible to do via WebBluetooth (see https://github.com/WebBluetoothCG/registries/blob/master/gatt_blocklist.txt ).

Is there an ETA by now (since it seems to be on the roadmap) or maybe even work in progress-branch we could support in for this bootloader to do the Nordic Secure OTA DFU? It would be greatly appreciated also if you clearly state that this is planned or not.

robsonos, 2023-12-25:
Hi,

I am also interested in this feature because of the WebBluetooth blocklist. @CSC-Sendance in the meantime, a good alternative is to use a different UUID for the DFU service.

Cheers.

fanoush, 2024-11-08:
HI guys. This bootlader + adafruit-nrfutil tool is basically an old legacy nordic bootloader + old nrfutil tool from SDK11 days. I think the whole point was to keep alive this simple unsigned legacy way of DFU as Nordic moved to the 'secure' = signed way of doing updates. So my question is - why don't you build the nordic DFU bootloader that is part of each SDK version from  SDK 12 and up as it looks like it is precisely targeted for you? The source is there and it is easy. And BTW you need to embed public key matching your private key into the bootloader source so you need to build your own version anyway.

SDKs are downloadable here https://developer.nordicsemi.com/nRF5_SDK/ , the one matching SoftDevice S140 6.1.1 is the 15.3.0 one and  dfu bootloader source is in examples/dfu/secure_bootloader so you just add your board header file for your hardware and generate signing key pair and  build it. The Makefime with list of sources needed is e.g. in  examples\dfu\secure_bootloader\pca10056_ble\armgcc\ if you would use gcc for building it.

And BTW the Espruino DFU bootloader is based on this newer 'secure' OTA way - mainly  because it uses SDK12 (for 52832) or 15 (for 52840) and it was easier to disable  the signature checks there then porting old legacy bootloader to SDK12 and up (which is what Adafruit did with this bootloader). So if for some reason building from Nordic SDK bootloader source seems tricky (it is not) then you can also checkout Espruino and build just the bootloader, the source used on top of SDK libraries is here https://github.com/espruino/Espruino/tree/master/targets/nrf5x_dfu and here https://github.com/espruino/Espruino/tree/master/targetlibs/nrf5x/components/libraries/bootloader

Also here is my patch for Nordic secure bootloader for SDK14 (matching softdevice S132 5.x) with signarure checks and board specific stuff removed and also with usage of optional 32kHz crystal disabled https://gist.github.com/fanoush/3420b9a3bccaaff3432ecf97f707958d just to get the idea.

In theory it is possible to add this newer DFU OTA implementation to this bootloader but what would you gain by doing it? It would be what the Nordic bootloader example source (+ nordic nrfutil tool) already is. And other features of this bootloader like uf2 is unsecure too, so it would only break the security for you.

fanoush, 2026-01-16:
@ejtagle looks like you went full circle with this solution :-) Nordic moved to secure DFU in SDK12 and SoftDevice => 3.0.0, Adafruit wanted to use new Softdevice 6.x (when moving from 52832 to 52840?) and so they backported SDK11 legacy bootloader so old tools could run with newer softdevice and now you implemented encryption and disabling of uf2 = basically negating (almost?) everything why this bootloader exists :-)

And it is also incompatible with Nordic secure bootloader and does not solve the issue mentioned above that web bluetooth blocklisted legacy bootloader guids.

I'd think it would be better and more natural to simply use newer nordic secure bootloader implementation instead of this legacy one if you need the secure DFU (see my post above). I wonder why using Nordic secure DFU bootloader source was not an option for you and what is the advantage of this solution?

But it is definitely great feature for someone who is stuck in project based on old  SDK11 and S132 softdevice version 2 and wants to quickly secure OTA updates without upgrading SDK and rewriting lot of the code.  Which is not the case here so it feels a bit out place and somewhat roundabout solution to the problem.