adafruit/Adafruit_nRF52_Bootloader #46: Copying UF2 fails on Windows 7 and 10

dhalbert, 2019-02-20:
[Originally posted by @ATMakersBill as https://github.com/adafruit/circuitpython/issues/1575.]

Hi folks, I have a Feather NRF52840 and tried to install CP 4.0b2 on it.  I entered bootloader mode and got the FTHR840BOOT drive.  I copied the UF2 and got a failure
[image]

I have gotten different errors saying that the file is too large to fit on the device, but it always fails.

Here is my bootloader version.
```
UF2 Bootloader 0.2.6 lib/nrfx (v1.1.0-1-g096e770) lib/tinyusb (legacy-525-ga1c59649) s140 6.1.1
Model: Adafruit Feather nRF52840 Express
Board-ID: NRF52-Bluefruit-v0
Bootloader: s140 6.1.1
Date: Dec 21 2018
```
After multiple attempts, I was able to get the REPL started on beta1.  I tried erase_filesystem() and it made no difference.

I switched to a second device and started with just beta1.  That gave me an odd error saying the file had not been copied... but it then started up CIRCUITPY and works.

I still have the broken device (which is still broken) if you want any testing done.

This has been replicated on Discord

dhalbert, 2019-02-20:
This is on Windows 10, 1809 build. I got a different error but also failed to copy the .uf2 to the Feather 52840.
```
UF2 Bootloader 0.2.6-10-g641d9fe lib/nrfx (v1.1.0-1-g096e770) lib/tinyusb (legacy-525-ga1c59649) s140 6.1.1
Model: Adafruit Feather nRF52840 Express
Board-ID: NRF52-Bluefruit-v0
Bootloader: s140 6.1.1
Date: Jan 31 2019
```

[image]

Upgraded to bootloader 0.2.8, though it reports as:
```
UF2 Bootloader 0.2.6-11-gd9b275c lib/nrfx (v1.1.0-1-g096e770) lib/tinyusb (legacy-525-ga1c59649) s140 6.1.1
Model: Adafruit Feather nRF52840 Express
Board-ID: NRF52-Bluefruit-v0
Bootloader: s140 6.1.1
Date: Feb 19 2019
```
Then I get the error shown in the original post, where the UF2 is reported to have disappeared from its original location.

dhalbert, 2019-02-20:
On Windows 7, trying to copy via drag and drop shows this error:
[image]

Works fine on Linux, btw.

hathach, 2019-02-20:
Thanks Dan for details. Weird, It used to work previously. I will upgrade my windows 10 to test again.

dhalbert, 2019-02-20:
It might be related to the size of the UF2, maybe?

hathach, 2019-02-20:
Indeed, on my windows 10, I got file **"too large for the destination file system"**. I am working on this

hathach, 2019-02-20:
more update: tested with arduino bleuart example and cpy 4.0.0 seems to work fine, despite error message when my pca10056 reboot to application mode

[image]

hathach, 2019-02-20:
@dhalbert  I tried to build the cpy master to test but got into this issue

```
GEN build-pca10056/autogen_display_resources.c
Traceback (most recent call last):
  File "../../tools/gen_display_resources.py", line 10, in <module>
    from adafruit_bitmap_font import bitmap_font
ModuleNotFoundError: No module named 'adafruit_bitmap_font'
../../supervisor/supervisor.mk:108: recipe for target 'build-pca10056/autogen_display_resources.c' failed
make: *** [build-pca10056/autogen_display_resources.c] Error 1
```

dhalbert, 2019-02-20:
Did you `git submodule update --init --recursive` ?

Here is the latest pca10056 build to save you testing time: 
[pca10056.zip](https://github.com/adafruit/Adafruit_nRF52_Bootloader/files/2885237/pca10056.zip)

hathach, 2019-02-20:
> Did you `git submodule update --init --recursive` ?
> 
> Here is the latest pca10056 build to save you testing time:
> [pca10056.zip](https://github.com/adafruit/Adafruit_nRF52_Bootloader/files/2885237/pca10056.zip)

Ah hah, thanks, I did run git submodule update (but forgot --init :D ). build now, and the bootloader still works as expected. @dhalbert I only have pca10056 here (no feahter nrf52840), can you confirm this issue is reproducable on pca10056 as well ?

dhalbert, 2019-02-20:
On pca10056 with the firmware.uf2 above, I got several of these errors, including the the "This is no longer located" error. I am running Windows 10 1809, which is the next 6-month update after what you 
have. Here is an example where it gets stuck copying.
[image]
What seems to happen is that it switches to CIRCUITPY prematurely, before Windows thinks the copy has completed. Could this be due to your noticing that the blocks you want to flash are already identical to what is coming in, and somehow aborting the rest of the copy?

If you can upgrade to 1809, you might get more info. However, it's a couple of GB upload and will take a while to install. 1803 or Windows 7 might also fail somehow. Try switching back and forth between different UF2's, maybe.

dhalbert, 2019-02-20:
What also seems to happen is that firmware.uf2 (or at least a part of it) ends up in CIRCUITPY, at least some of the time.

hathach, 2019-02-20:
Thanks Dan for confirming this, I tried to hit update button several times but it keeps saying I am latest (maybe it is regional differences). I will try to see if I could manually got it to 18.09 to test. One of the uf2 implementation I remembered is ignored writing non-uf2 and assume OS will cache all of that information without verification. Maybe latest windows create some .cache file that It did check. Not sure though, I will get it update first. 

For pre-mature reset, the bootloader should only reset once it confirmed that the whole data is written. So it should not be an issue, I was thinking maybe windows did verify what it writes to uf2 block, and since the data isn't written yet (flashing) the readback is diff to written which trigger some windows disk cache issue of some sort.

dhalbert, 2019-02-20:
I think after Windows writes the file it will attempt to update the metadata in the FAT table and the directory entry. Maybe the bootloader is not waiting long enough for that to happen.

The error about the file disappearing on the Windows side is bizarre, and maybe that's a Windows bug they fixed in 1809.

This might be helpful about forcing an upgrade: https://support.microsoft.com/en-us/help/4028685/windows-10-get-the-update. I can't find a list of non-English 1809 releases and whether some aren't available yet.

hathach, 2019-02-20:
@dhalbert thanks again for the link, I just figured that out a couple minutes ago and is downloading the update, waiting for it to complete though. Luckily I used English version :)

hathach, 2019-02-21:
take a few hours to update to 18.09 and I could able to reproduce the issue, troubleshooting it now.

commit 9ed3f5ea (2019-02-21): fix #46 complete uf2 writing using tud_msc_write10_complete_cb
commit 7df44a54 (2019-02-22): Merge pull request #47 from adafruit/develop

fix #46 issue with windows 10 build 18.09
closed 2019-02-22 by dhalbert