adafruit/Adafruit_nRF52_Bootloader #48: 1.8Volts on  PCA10059 ("Dongle")

TasosR83, 2019-02-28:
After flashing the bootloader to the "Dongle", the dongle's on-chip LDO regulator gets to the mode where he outputs 1.8Volts instead of the standard (when yoou buy it) of 3.3Volts.

This makes the device  un-programable even with the j-link EDU (can be programmed with only the J-link PRO)

jerryneedell, 2019-02-28:
are you referring to the J-Link Mini https://www.adafruit.com/product/3571 or the "full sized Link- EDU 
 https://www.adafruit.com/product/1369

TasosR83, 2019-02-28:
the 

> are you referring to the J-Link Mini https://www.adafruit.com/product/3571 or the "full sized Link- EDU
> https://www.adafruit.com/product/1369

The 2nd one (SEGGER J-Link EDU) not the mini

jerryneedell, 2019-02-28:
hmmm -- I have used mine to program the SD and UF2 Bootloader to the Dongle -- I guess I have not gone beyond that with it.  I was not ware o f any difference between the EDU and PRO version other than terms of use.

jpconstantineau, 2019-02-28:
I think that setting it up in the bootloader would make a lot of sense.
Running at 1.8V makes the LEDs not work.
It took ma a while to figure out that I had to run this code to switch it
over:

if (NRF_UICR->REGOUT0 != UICR_REGOUT0_VOUT_3V3)
{
    NRF_NVMC->CONFIG = NVMC_CONFIG_WEN_Wen << NVMC_CONFIG_WEN_Pos;
    while (NRF_NVMC->READY == NVMC_READY_READY_Busy){}
    NRF_UICR->REGOUT0 = UICR_REGOUT0_VOUT_3V3;

    NRF_NVMC->CONFIG = NVMC_CONFIG_WEN_Ren << NVMC_CONFIG_WEN_Pos;
    while (NRF_NVMC->READY == NVMC_READY_READY_Busy){}
NRF_USBD->ENABLE = 1;
            delay(500);
        NVIC_SystemReset();
}

TasosR83, 2019-02-28:
> I think that setting it up in the bootloader would make a lot of sense. Running at 1.8V makes the LEDs not work. It took ma a while to figure out that I had to run this code to switch it over: if (NRF_UICR->REGOUT0 != UICR_REGOUT0_VOUT_3V3) { NRF_NVMC->CONFIG = NVMC_CONFIG_WEN_Wen << NVMC_CONFIG_WEN_Pos; while (NRF_NVMC->READY == NVMC_READY_READY_Busy){} NRF_UICR->REGOUT0 = UICR_REGOUT0_VOUT_3V3; NRF_NVMC->CONFIG = NVMC_CONFIG_WEN_Ren << NVMC_CONFIG_WEN_Pos; while (NRF_NVMC->READY == NVMC_READY_READY_Busy){} NRF_USBD->ENABLE = 1; delay(500); NVIC_SystemReset(); }
> […](#)
> -- Pierre

Can you please help me on how to build this code? I think you are referring to the Keil  (Keil Embedded Development Tools) and not to device programming using Arduino IDE, right ?

jerryneedell, 2019-02-28:
ah -- this was discussed here -- and there are notes about using the J-Link with it.
it is reset to 3.3V  fixed in the CircuitPython code.

dhalbert, 2019-02-28:
The voltage fix is in: https://github.com/adafruit/circuitpython/blob/master/ports/nrf/peripherals/nrf/nrf52840/power.c. Note that you have to do a hard reset if the voltage is wrong.

TasosR83, 2019-03-08:
Case should be closed. Thanks for your help. The easiest way to fix the wrong Voltage is the command:

`nrfjprog --memwr 0x10001304 --val 0xFFFFFFFD`

which will turn the internal regulator's output to 3.3 Volts. Maybe any device reset by power cycling the device is needed. 
Also to run the above you should use an SWD programmer (e.g. JLink EDU). 

If your programmer cannot 'speak' with 1.8V devices, use a power supply and give 3.3Volts to the VDD OUT pin (look [here](https://devzone.nordicsemi.com/f/nordic-q-a/44184/low-voltage-1-8v-on-nrf52840-cannot-program-chip/173413#173413)).

closed 2019-03-08 by TasosR83