adafruit/Adafruit_nRF52_Bootloader #50: USB Dongle (Nordic's) won't boot with 5V without connection with PC

TasosR83, 2019-03-07:
I am prototyping something using the nRf52840 USB Dongle.

Nordic's Dongle works with the power supply from the USB
BUT doesn't boot on the following  2 cases:
1) by applying power to the VBUS and GND even-though those two pins are connected to the power supply of the USB port.
2) by connecting it o a USB hub that provided only power (not data). I used an external-powered USB-hub and the dongle started working when the hub was connected to the PC, but didn't start working when the hub wasn't connected to the PC, but connected to the power.

I suspect this fault is caused by the bootloader and not the Nordic's Dongle. Correct me if I am wrong

henrygab, 2019-03-11:
Can you help me clarify the issue?

You wrote you are using "Nordic's Dongle".  Can you confirm that this refers to **PCA10059**?

Also, in your edit your wrote:
> EDIT:
> Is it a problem that happens only with Adafruit's bootloader, it
> oesnt happen with the bootloader that Nordic provides

Can you confirm that the first two words are swapped, and that you intended to indicate that this problem only occurs when using the Adafruit bootloader?

A final clarification that will be helpful is the version of the bootloader you are using.

_Note: I'm not part of Adafruit or this repo, just a user...._

TasosR83, 2019-03-12:
> Can you help me clarify the issue?
>
> You wrote you are using "Nordic's Dongle". Can you confirm that this refers to **PCA10059**?
>
> Also, in your edit your wrote:
>
>
> Can you confirm that the first two [...]

Thanks a lot for your comments, I updated the original Issue

samnaughtonb, 2019-03-20:
If you don't mind me asking, how did you flash the Adafruit bootloader in the first place? I've been trying (on OSX) and can't seem to get it to work. Also, other resources on the internet seem to require the use of a J-Link (which I don't have).
Thanks

TasosR83, 2019-03-21:
> If you don't mind me asking, how did you flash the Adafruit bootloader in the first place? I've been trying (on OSX) and can't seem to get it to work. Also, other resources on the internet seem to require the use of a J-Link (which I don't have).
> Thanks

Unfortunately for you, I used the SWD interface of a Segger Jlink EDU (the one that costs about 60euros, not the "mini" version), but  [some people](https://forum.mysensors.org/topic/9717/everything-nrf52840/116) have found other cheaper programmers. The Programmer application of the "nrf for desktop" program, wont let you change the bootloader if it is not signed from Nordic ('unsigned bootloader' meassage)

hathach, 2020-03-24:
closed due to lack of activities and this is not an issue of this repo

closed 2020-03-24 by hathach

henrygab, 2020-03-24:
@hathach -- Can you help me understand why this issue would not be considered an issue of this repo?

The user indicated it occurred when using an _Adafruit_ bootloader from this depot's [releases](https://github.com/adafruit/Adafruit_nRF52_Bootloader/releases).  I'm sure I'm just missing something obvious, so thanks for helping me figure out what that would be! :)

hathach, 2020-03-25:
@henrygab ah you are right, it is my bad. I forgot what is this issue about. Have a brief look at the 1st port, and thought it is the hardware issue. Didn't read to the line stating it only happens with this repo. However, since pca10059 board is not officially supported, I won't spend time troubleshooting this, user can re-open or file an new issue when having more activities.