adafruit/Adafruit_nRF52_Bootloader #51: UF2: mismatched media types

henrygab, 2019-03-11:
### Issue:
From Microsoft [FATGEN103.pdf](https://staff.washington.edu/dittrich/misc/fatgen103.pdf): _"Whatever is placed [in boot sector media descriptor] must also be put in the low byte of the FAT[0] entry"_

From the UF2 exposed on a feather, using feather_nrf52840_express_bootloader-0.2.9_s140_6.1.1.hex:

Struct/Field | Value | Start | Length | Note
-- | -- | -- | -- | -- 
boot_sector.media_descriptor | F8h | 15h | 1h | Hard drive
boot_sector.drive_number | 0h | 24h | 1h  | Corresponds to floppy
fat0.media_type | FFF0h | 200h | 2h | Floppy
fat1.media_type | FFF0h | 4200h | 2h | Floppy

### Fix
PR request coming soon.

hathach, 2019-03-11:
Thank you very much @henrygab , according to the above doc. MS-DOS v2 will only look at the BPB and skip the first byte in the FAT structure. But I agree they should match the other.

> The determination of which type was on the disk was done by looking at the first byte of
the FAT (the low 8 bits of FAT[0]).
This type of media determination was superseded in MS-DOS version 2.x by putting a BPB in the
boot sector, and the old style of media determination (done by looking at the first byte of the FAT)
was no longer supported. All FAT volumes must have a BPB in the boot sector.

hathach, 2019-03-13:
fix by PR #52

closed 2019-03-13 by hathach