adafruit/Adafruit_nRF52_Bootloader #58: Add UF2 build/generation information to Readme

Spegs21, 2019-03-25:
The only Adafruit supplied information about how to build for UF2 that I could find is found [here](https://learn.adafruit.com/adafruit-feather-m0-express-designed-for-circuit-python-circuitpython/uf2-bootloader-details).

It says to start to the flash at 0x4000 for M4 boards, which this is. With the MBR and soft device that doesn't work. The flash needs to be placed at 0x26000 with a flash size of 0xda000. The example blinky_pca10059_mbr works fine when updating through UF2  just by changing this in the Section Placement Macros in SES.

Additionally, I had to dig through the issues to find the family code, 0xADA52840.

This information should be more easily accessible.

ladyada, 2019-03-25:
thanks for the note, if you can submit a PR with any relevant info you think we should point out - we'll take a look :)

Spegs21, 2019-03-28:
Alright, I put in the PR. It took me an hour to figure this out. I'm just hoping this will make it easier for someone else in the future.

closed 2019-03-28 by Spegs21

ladyada, 2019-03-28:
great work - thanks for the contribution!