adafruit/Adafruit_nRF52_Bootloader #59: nRF52840 Feather Express Pinout problem

normra, 2019-03-25:
Symptom: connect SPI device to MISO pin and restart Feather.   Bootloader enters Mass Storage DFU mode.

Problem: In src/boards/feather_nrf52840_express.h, the following appears
```
#define BUTTON_1              _PINNUM(0, 15)
#define BUTTON_2              _PINNUM(0, 19)
```

This does not agree with Adafruit [schematic ](https://cdn-learn.adafruit.com/assets/assets/000/068/545/original/circuitpython_nRF52840_Schematic_REV-D.png?1546364754) which appears to require
```
#define BUTTON_1              _PINNUM(1, 02)
#define BUTTON_2              _PINNUM(0, 18)
```

hathach, 2019-03-27:
thank you very much, it is indeed an issue. We change schematic at the end of prototyping and miss this one.Button 2 is mapped to D2 though

closed 2019-03-27 by hathach

normra, 2019-03-27:
I have a question about your comment that 'Button 2 is mapped to D2 though'.   Are you saying the published schematic is wrong?  On the released version referenced above, Button 2 (labelled SW1) is connected via line labelled RESET to physical pin 40 of the module which is labelled P0.18/RESET.

hathach, 2019-03-27:
@normra sorry for not being clear enough. The published schematic is absolute `correct`, D2 is pin that is label in the PCB, which is next to TX and is P0.10
https://learn.adafruit.com/introducing-the-adafruit-nrf52840-feather/overview

Thank you again.

normra, 2019-03-27:
Sorry to be thick here -  I agree that D2 is P0.10.  However, on schematic, the Reset button (aka SW1 aka Button 2) isn't connected to D2.   It's connected to P0.18/RESET.

Here is extract from published schematic 

[image]

Apologies in advance if I'm misreading this.

normra, 2019-03-28:
I am misreading this.  BUTTON2 is FRESET not RESET.   And it's not connected to SW1 - I need a jumper to assert this signal.

Sorry for confusion.

hathach, 2019-03-28:
no problem, thank you for your issue :)

commit 27130696 (2019-03-27): fix #59 incorrect button mapping for feather nrf52840
commit 60debfc2 (2019-03-27): Merge pull request #60 from adafruit/develop

fix #59 incorrect button mapping for feather nrf52840