adafruit/Adafruit_nRF52_Bootloader #72: Upgrade to support Softdevice ver 7.0.0

hathach, 2019-06-25:


dhalbert, 2019-06-25:
I'm waiting for the doc of the changes, and would probably wait for at least 7.0.1 so they can iron out bugs. If there are upward-compatibility issues we have to be careful.

hathach, 2019-06-26:
agree, we should wait for the next release with bug fix. I will make branch to start adopt the new flash layout and testing.

hathach, 2019-11-28:
@dhalbert SD v7 seems to be required to support the new nrf52833 and probably nrf5340 later on as well. 
https://www.nordicsemi.com/Software-and-tools/Software/S140/Download#infotabs
[image]

dhalbert, 2019-11-29:
Thanks. We can discuss about an upgrade strategy next week.

hathach, 2019-11-29:
@dhalbert Please take the break, It is holiday in USA. I already wrapped it up with #88 . We will upgrade the SD later when **it supports 5340 dual core chip**. Since we probably make a feather for 5340 than 52833 :) .

PS: Next week I will take a whole week off for a vacation. Though I am still reachable via email (hopefully)

dhalbert, 2020-08-15:
@hathach Do we need to upgrade everything to 7.x.x, or can 6.x.x and 7.x.x coexist? I was going to build this for pca10100 just so I could use it to upload the SD on a pca10100 board but realized it was not updated yet. Thanks.

hathach, 2020-08-17:
@dhalbert it can be co-existed, the bootloader can work with either bootloader since they are decoupled. However, since the flash layout changes due to SD_SIZE, the application must has correct linker. Actually, I would suggest to include the Softdevice as part of the application uf2, although the uf2 is larger, it doesn't actually take time to flash since most of the time the softdevice matches with the flash contents. To sum up

- You can flash bootloader with/wihtout SD v6 as current state of bootloader
-  In both ways, just include SD v7 (or any later SD version) as part of circuitpython to uf2 (use mergehex before uf2conv), it should then just work for all the board.

Note: This need latest bootloader (master tip) which isn't released yet since I think we need a bit more testing. Hope this help.

pabigot, 2021-04-12:
I was looking into this a little: coexisting for S132 through 7.2.0 should be fine, but S140 (nRF52840) at 7.0.0 increased the required flash space from 0x26000 to 0x27000, which from what I can tell would require changing the bootloader and application linker maps.

For now I'm just sticking with 6.1.1.

closed 2021-11-29 by hathach