adafruit/Adafruit_nRF52_Bootloader #73: Support for the S340 (ANT + BLE) SoftDevice

judepereira, 2019-07-10:
**Is your feature request related to a problem? Please describe.**
Not a problem as such, but a blocked to write ANT and ANT+ code.

**Describe the solution you'd like**
Support for the S340 (ANT + BLE) SoftDevice. This SoftDevice profile must be downloaded from thisisant.com, and thus the hex file cannot be shared. However, apart from that, the linker script needs to be created.

The blocker for me is this linker script, as I have no idea how to get the values for `BOOTLOADER_SETTINGS` and `MBR_PARAMS_PAGE`.

**Describe alternatives you've considered**
No alternatives.

**Additional context**
This is the only SoftDevice which supports ANT for the nRF52840 series.

ladyada, 2019-07-10:
Hi, if you are using Sparkfun Mini nRF52840 please ask Sparkfun to submit a PR for this technology

judepereira, 2019-07-10:
@ladyada I believe this should be independent of the board, since the linker script uses values from within the SoftDevice hex.

ladyada, 2019-07-10:
For hardware purchased from Sparkfun, they will be able to help you with your bootloader questions and they can submit a PR

closed 2019-07-10 by ladyada

judepereira, 2019-07-11:
@ladyada Just as an FYI, the steps do not differ for adding support for the s340 SoftDevice to the SparkFun or the Adafruit Feather. Here's proof: https://judepereira.com/blog/nrf52840-flash-s340/

Nothing SparkFun related.

I'd gladly open up a PR, but it looks like anything even remotely connected to SparkFun will be gladly overlooked and passed up on.

I'm deeply thankful for Adafruit for open sourcing their bootloader.

orrmany, 2019-12-30:
I would also need it for the Adafruit nrf52840 Express feather. I would like to use it for ANT stuff