adafruit/Adafruit_nRF52_Bootloader #74: Use without softdevice

antst, 2019-08-04:
Hello,

I am trying to adopt this bootloader for use in non-BLE application, without softdevice.
Mainly due to UF2. 

By commenting out -DBLE_STACK_SUPPORT_REQD and -DSOFTDEVICE_PRESENT and excluding compilation of few BLE-specific binaries and wrapping in ifdef few BLE-specific calls I was able to compile it, but it doesn't work for me. (I test on nrf52840-DK with flashed MBR)
Looking that there are those defines, it looks like somebody had in mind use of this boot-loader in non-BLE application. 
Were there attempts to run without softdevice? Any suggestions?

Thanks!

xiongyihui, 2019-12-12:
+1
want to use it with zephyr or mynewt which has its own ble stack

hathach, 2019-12-12:
related to #30

hathach, 2020-05-29:
implemented by #128

closed 2020-05-29 by hathach