adafruit/Adafruit_nRF52_Bootloader #85: Entering on OTA for nRF52840 Express seems not working

christophepersoz, 2019-10-27:
**Describe the bug**
After reading the code and the `readme.md`, entering in OTA required `DFU = LOW` and `FRST=LOW`. 
In`/src/boards/feather_nRF52840_express/board.h` there is this description,

```
/*------------------------------------------------------------------*/
/* BUTTON
 *------------------------------------------------------------------*/
#define BUTTONS_NUMBER        2
#define BUTTON_1              _PINNUM(1, 02)
#define BUTTON_2              _PINNUM(0, 10)
#define BUTTON_PULL           NRF_GPIO_PIN_PULLUP
```

this `defined` are then included into `/src/board.h` as follow,
```
#include "board.h"

#ifndef BUTTON_DFU
#define BUTTON_DFU      BUTTON_1
#endif

#ifndef BUTTON_FRESET
#define BUTTON_FRESET   BUTTON_2
#endif
```

But, `BUTTON_2` have no correspondance the the schematic found there : [nRF52840 Schematic](https://cdn-learn.adafruit.com/assets/assets/000/068/545/original/circuitpython_nRF52840_Schematic_REV-D.png?1546364754)
Only `BUTTON_1` match to `_PINNUM(1,02)` as `SWITCH` on schematic.

**Set up (please complete the following information)**
  - Board : Adafruit nRF52840 Express
  - Mac OS 10.14.6

**To Reproduce**
1. Just try to enable OTA with `BUTTON_1` and `BUTTON_2`, it won't work

**Expected behavior**
Have the OTA enabled as expected by pressing `BUTTON_1` and `BUTTON_2`

**Additional context**
To solve this, simply replace in `/src/boards/feather_nRF52840_express/board.h` 
```
#define BUTTON_1              _PINNUM(1, 02)
```
by
```
#define BUTTON_1              _PINNUM(0, 18)
```

JoeBakalor, 2019-12-17:
@christophepersoz We added triple reset to enter OTA on our fork if you want to take a look https://github.com/adafruit/Adafruit_nRF52_Bootloader/pull/91

christophepersoz, 2019-12-22:
Thanks a lot for your fork link @JoeBakalor, I'm gonna have a look at it!

hathach, 2019-12-23:
Yeah, it is not so easy now requiring GND 2 pins. I will discuss with others to see what is the best way to enter OTA. However, to be honest I don't really like the triple reset method.

christophepersoz, 2019-12-23:
Agree too @hathach, for the moment I changed the bootloader to have the following sequence to enter OTA DFU, User BT down on power-up. 
The best could be, this User Switch hold for a second after a RESET or a power-up cycle to enter in, but I haven’t explore this way yet.

ElreboCM, 2021-03-02:
There seems to be the same issue for the Adafruit nRF52840 Feather Sense. 
For me even connecting D2 to ground won't work
I am currently powering through USB. Could this be the causing the problem?

ElreboCM, 2021-03-03:
I can confirm the process described by @christophepersoz as working. 
I was on Bootloader 0.2.9 with the sense. After upgrading to 0.3.1 it was working ( also tested on 0.4.1).

hathach, 2021-03-04:
I am currently focusing on other work. I will try to look at this later on when working back with bootloader later on.