adafruit/Adafruit_nRF52_Bootloader #94: Flashing on Arduino Nano 33 BLE?

rpavlik, 2020-01-03:
I see it's listed, and it builds, but adafruit-nrfutil doesn't work, and my simple bossac command didn't work either. (have used the reset button to get into bootloader, but still no dice) Do I need to find the swd pins? or is there a way to flash it e.g. like the .ino update-bootloaders for the samdx1 boards?

My command line was:

```
~/.arduino15/packages/arduino/tools/bossac/1.9.1-arduino1/bossac -d --port=ttyACM0 -b1 -U -i -e -w  _build-arduino_nano_33_ble/arduino_nano_33_ble_bootloader-0.2.13-19-ge25b63a_s140_6.1.1.hex -R
```

Didn't matter which hex I picked (with or without sd) neither of them appeared to actually have an effect, despite seeming to complete a flash process, I'd still get no MSC device and bossac showed the same ID.

ladyada, 2020-01-03:
did you install the bootloader with a jlink as per
https://learn.adafruit.com/circuitpython-on-the-nrf52/nrf52840-bootloader

rpavlik, 2020-01-03:
No, I was hoping to not need a j-link, just have the bossac/sam-ba Arduino bootloader that it came with on it right now. I've only got stlinkv2, atmel-ice, and black magic probe, so looks like I've got the start of a new Adafruit order already with a j-link mini edu 😉

ladyada, 2020-01-03:
black magic might be able to do it - but we have not tried it ourselves so we dont know the exact command to run

rpavlik, 2020-01-03:
Ok thanks! I'll poke around and let you know if I figure it out with the bmp.

rpavlik, 2020-01-06:
OK, so the BMP can do it too!

Pinout of bottom port is: 

- Reset  -- GND
- no pad -- SWCLK/TCLK
- SWDIO/TMS -- VCC (for target voltage detect)

(see https://www.hackster.io/visualmicro/arduino-nano-33-iot-debugging-633ad8 - couldn't find a license ref for the arduino store's photo)

Assuming you've done the [`/etc/udev/rules.d/99-blackmagic.rules` file](https://github.com/blacksphere/blackmagic/wiki/Frequently-Asked-Questions#its-annoying-to-look-up-an-always-changing-device-name-on-linux) to get a `/dev/ttyBmpGdb` symlink, the following flashes it: (I've put the commands into a single "one-liner" though I ran them separate. I could presumably put this in the makefile if you're interested):

```sh
gdb -nx --batch \
  -ex 'target extended-remote /dev/ttyBmpGdb' \
  -ex 'monitor swdp_scan' \
  -ex 'attach 1' \
  -ex 'load arduino_nano_33_ble_bootloader-0.2.13-19-ge25b63a_s140_6.1.1.hex' \
  -ex 'compare-sections' \
  -ex 'kill'
```

You'll get some warnings and stuff but they don't appear to affect it (too much, anyway) - it at least installs well enough to reboot and expose the nano33boot MSC device. Further flashes of the bootloader can be done as usual with `make SERIAL=/dev/ttyACM1 BOARD=arduino_nano_33_ble all combinehex dfu-flash`

However, it does disconnect/reconnect regularly - here's my dmesg with a 5.2 kernel

```
 [62476.400525] usb 3-1.2.2: New USB device found, idVendor=239a, idProduct=0063, bcdDevice= 1.00
[62476.400526] usb 3-1.2.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[62476.400528] usb 3-1.2.2: Product: Nano 33 BLE
[62476.400528] usb 3-1.2.2: Manufacturer: Arduino
[62476.400529] usb 3-1.2.2: SerialNumber: E3D4B66F87C02642
[62476.457501] cdc_acm 3-1.2.2:1.0: ttyACM0: USB ACM device
[62476.457923] usb-storage 3-1.2.2:1.2: USB Mass Storage device detected
[62476.458014] scsi host9: usb-storage 3-1.2.2:1.2
[62477.467545] scsi host9: scsi scan: INQUIRY result too short (5), using 36
[62477.467548] scsi 9:0:0:0: Direct-Access     Adafruit nRF UF2          1.0  PQ: 0 ANSI: 2
[62477.467760] sd 9:0:0:0: Attached scsi generic sg1 type 0
[62477.468072] sd 9:0:0:0: [sdb] 8000 512-byte logical blocks: (4.10 MB/3.91 MiB)
[62477.468283] sd 9:0:0:0: [sdb] Write Protect is off
[62477.468284] sd 9:0:0:0: [sdb] Mode Sense: 03 00 00 00
[62477.468480] sd 9:0:0:0: [sdb] No Caching mode page found
[62477.468482] sd 9:0:0:0: [sdb] Assuming drive cache: write through
[62477.477314]  sdb:
[62477.478675] sd 9:0:0:0: [sdb] Attached SCSI removable disk
[62664.137348] usb 3-1.2.2: USB disconnect, device number 87
[62664.146033] FAT-fs (sdb): unable to read boot sector to mark fs as dirty
[62668.202655] usb 3-1.2.3: new full-speed USB device number 88 using xhci_hcd
[62668.304452] usb 3-1.2.3: New USB device found, idVendor=239a, idProduct=0063, bcdDevice= 1.00
[62668.304454] usb 3-1.2.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[62668.304455] usb 3-1.2.3: Product: Nano 33 BLE
[62668.304456] usb 3-1.2.3: Manufacturer: Arduino
[62668.304457] usb 3-1.2.3: SerialNumber: E3D4B66F87C02642
[62668.361597] cdc_acm 3-1.2.3:1.0: ttyACM0: USB ACM device
[62668.362636] usb-storage 3-1.2.3:1.2: USB Mass Storage device detected
[62668.362807] scsi host9: usb-storage 3-1.2.3:1.2
[62669.371968] scsi host9: scsi scan: INQUIRY result too short (5), using 36
[62669.371972] scsi 9:0:0:0: Direct-Access     Adafruit nRF UF2          1.0  PQ: 0 ANSI: 2
[62669.372237] sd 9:0:0:0: Attached scsi generic sg1 type 0
[62669.372508] sd 9:0:0:0: [sdb] 8000 512-byte logical blocks: (4.10 MB/3.91 MiB)
[62669.372722] sd 9:0:0:0: [sdb] Write Protect is off
[62669.372723] sd 9:0:0:0: [sdb] Mode Sense: 03 00 00 00
[62669.372933] sd 9:0:0:0: [sdb] No Caching mode page found
[62669.372937] sd 9:0:0:0: [sdb] Assuming drive cache: write through
[62669.382049]  sdb:
[62669.383378] sd 9:0:0:0: [sdb] Attached SCSI removable disk
[62968.521849] usb 3-1.2.3: USB disconnect, device number 88
[62968.531270] FAT-fs (sdb): unable to read boot sector to mark fs as dirty
[62968.837989] usb 3-1.2.3: new full-speed USB device number 89 using xhci_hcd
[62968.943450] usb 3-1.2.3: New USB device found, idVendor=239a, idProduct=0063, bcdDevice= 1.00
[... 119 lines cut ...]
PORTS {/dev/ttyACM0, } / {/dev/ttyACM0, } => {}
PORTS {/dev/ttyACM0, } / {/dev/ttyACM0, } => {}
PORTS {/dev/ttyACM0, } / {/dev/ttyACM0, } => {}
PORTS {/dev/ttyACM0, } / {/dev/ttyACM0, } => {}
PORTS {/dev/ttyACM0, } / {/dev/ttyACM0, } => {}
PORTS {/dev/ttyACM0, } / {/dev/ttyACM0, } => {}
PORTS {/dev/ttyACM0, } / {/dev/ttyACM0, } => {}
PORTS {/dev/ttyACM0, } / {/dev/ttyACM0, } => {}
PORTS {/dev/ttyACM0, } / {/dev/ttyACM0, } => {}
PORTS {/dev/ttyACM0, } / {/dev/ttyACM0, } => {}
PORTS {/dev/ttyACM0, } / {/dev/ttyACM0, } => {}
PORTS {/dev/ttyACM0, } / {/dev/ttyACM0, } => {}
PORTS {/dev/ttyACM0, } / {/dev/ttyACM0, } => {}
PORTS {/dev/ttyACM0, } / {/dev/ttyACM0, } => {}
PORTS {/dev/ttyACM0, } / {/dev/ttyACM0, } => {}
Uploading using selected port: /dev/ttyACM0
/home/ryan/.arduino15/packages/arduino/tools/bossac/1.9.1-arduino1/bossac -d --port=ttyACM0 -U -i -e -w /tmp/arduino_build_949974/Blink.ino.bin -R 
No device found on ttyACM0
Set binary mode
An error occurred while uploading the sketch
Send auto-baud
Set binary mode
```

I don't have any other uf2's built for nrf52840 to test, unfortunately.

rpavlik, 2020-01-06:
Oh, but if I call it an ItsyBitsy nrf52840, flashing "Blink" to it will blink the blue part of the RGB LED, which is something. Of course, your board defs use adafruit-nrfutil instead of bossac.

rpavlik, 2020-01-06:
OK, some of my small tweaks are up on https://github.com/adafruit/Adafruit_nRF52_Bootloader/pull/97 . Since I can't find the original bootloader's source code, and adding SAM-BA/bossac support would probably increase code size a bit, I think my next approach will be to make a modified board definition for Arduino to use nrfutil instead of bossac.

rpavlik, 2020-01-07:
OK, I see part of the issue with the UF2 part: The current.uf2 file is listed as only 512bytes (so 1 uf2 block) in size, compared to 598KB for my CPB. I get most of the same dmesg warnings on the CPB, so they're not related to my issues. The capacity of the "drive" is shown as the same (`8000 512-byte logical blocks: (4.10 MB/3.91 MiB)`), but for some reason the user portion of the flash isn't showing up right.  Not sure why - comparing with the itsybitsy '840 (similar pin count) and feather '832 (same lack of qspi flash), the board config files look similar.

On the plus side, the nrfutil-used DFU stuff works fine, so I've been able to try a handful of bluefruit sketches.

hathach, 2020-07-14:
closed due to lack of activities.

closed 2020-07-14 by hathach