adafruit/Adafruit_nRF52_Bootloader #98: Detailed build instructions

samveen, 2020-01-10:
Is it possible to add detailed build instructions and environment information for the bootloader?

**Background**
I am using an AdaFruit feather nRF52832 for a project, and I think that the softdevice s132 is overkill for my requirements, and I should be able to get by with either the softdevice s112 or the softdevice s113. Towards that end I was trying to figure out how to build a bootloader with the appropriate softdevice, but I don't see any details on how to set up the build environment for the bootloader.

ladyada, 2020-01-10:
we've got details here!
https://learn.adafruit.com/circuitpython-on-the-nrf52/nrf52840-bootloader

closed 2020-01-10 by ladyada

samveen, 2020-01-10:
> we've got details here!
> https://learn.adafruit.com/circuitpython-on-the-nrf52/nrf52840-bootloader
Pardon the initial title and requirements. The mobile experience of GitHub turned me stupid. I now see that the detailed build instructions are already present. This I have repurposed this issue for what I actually wanted to ask, instead of preliminaries.