adafruit/Adafruit_nRF52_Bootloader #101: Softdevice s112 or s113 builds for nRF52832

samveen, 2020-01-14:
I am using an AdaFruit feather nRF52832 for a project and, as I see it, softdevice s132 is overkill for my requirements. I should be able to get by with either the softdevice s112 or the softdevice s113.

Towards that end I have managed to to figure out how to build the bootloader with the appropriate softdevice (both s112_6.1.1, and s113_7.0.1) . 
What I don't have is the linking instructions to combine the output of the build with the softdevice hex files to build a viable bootloader.

Can I get this information from somewhere, or if possible can I get an appropriate ld file for this?

I have considered using the linker file for s132_6.1.1, but I am pretty sure that's not going to work without the required information that I am missing already.

My work is available at samveen/Adafruit_nRF52_Bootloader#1

hathach, 2020-01-14:
Thank you for sharing your work with us. However, I don't think we have enough resource to support, maintain and test multiple SoftDevice especially their flash layout can be difference which make application incompatible between SD. And I don't see much advantages by using s112/s113 over s132/s140. You are pretty on your own.

@dhalbert @ladyada

ladyada, 2020-01-14:
yes agreed - changing SD is not trivial and has huge repercussions!

closed 2020-01-14 by ladyada

samveen, 2020-01-15:
@ladyada @hathach  I am not expecting support, just some pointers as to where I could find some of the information used in the repo here, specifically the linker information

ladyada, 2020-01-15:
linker files are here https://github.com/adafruit/Adafruit_nRF52_Bootloader/tree/master/src/linker

hathach, 2020-01-15:
@samveen words of warning: the bootloader code may has some magic number regarding the application start address, be sure to check it out. Note: if the S112 flash layout is not the same as S132, you may not (or may) use our Arduino and/or Cpython repo.

samveen, 2020-01-22:
@hathach What I was hoping for from you was the source from which you derived the linker files for the nrf52832 repo. I have looked at the nRF sdk (16.0.0) boot loader examples for softdevice s132 and s140, but I don't see anything obvious in the comparison. If you can give me any other sources/docs that you remember you read, that would hopefully help. 

For testing purposes, I will be forking the arduino core too and getting it stripped down to match the s112/s113 bootloader in an effort to test the bootloader.

gabewillen, 2021-01-09:
> @hathach What I was hoping for from you was the source from which you derived the linker files for the nrf52832 (the source where you got the flash layout from). I have looked at the nRF sdk (16.0.0) boot loader examples for softdevice s132 and s140, but I don't see anything obvious in the comparison. If you can give me any other sources/docs that you remember you read, that would hopefully help.
> 
> For testing purposes, I will be forking the arduino core too and getting it stripped down to match the s112/s113 bootloader in an effort to test the bootloader.

Did you get this to work?