adafruit/Adafruit_nRF52_Bootloader #105: Makefile compile error

thaanstad, 2020-01-27:
I am having issues compiling the bootloader for the NRF52840 feather express (per the example) using the makefile on a windows 8.1 machine.  I was interested in using segger embedded studio for the build but per the BUG #89 there appear to be some issues with that and I encountered similar problems. 

To build with the make file I first installed ARM GCC and Nordic Command line tools and then MinGW to get the make command and then GIT to allow for the git command to work.  If necessary I added the programs to the windows PATH and verified I had access to them from the command line.  I now have an error that says mkdir: command not found as well as make: *** [_build-feather_nrf52840_express] Error 127.  When I enable verbose output I get another error before the others: make: /bin/sh: command not found.  Am I missing another program to allow the make file to compile the bootloader on a windows machine or is there an issue with the makefile and windows?  The makefile instructions didn't mention another program but perhaps it is easier on a linux machine?  I can test mkdir in command line and it is able to make directories.

**Hardware
  - Custom board based on Adafruit NRF52 feather
  - Windows 8.1

**To Reproduce**
make BOARD=feather_nrf52840_express all combinehex V=1

**Expected behavior**
The bootloader compiles and exports a .hex file for uploading with nrfGo studio or arduino.

**Screenshots**
[image]

FYI:
I made a custom PCB based off the NRF52832 feather using the miniature Taiyo Yuden version of the NRF52 and mixed up the RX and TX pins so I want to modify the bootloader to change the pins.  I verified the board works with the original bootloader if I make the connections with jumpers but this is difficult and I figured I could try compiling a bootloader as well!

ladyada, 2020-01-27:
you may need to build in a linux environment. please note we don't provide tech support for DIY builds, we build this on win10 all the time with a good compilation environment so its a setup issue which we cannot debug for you :)

closed 2020-01-27 by ladyada

thaanstad, 2020-01-27:
I believe the error regarding make: /bin/sh was due to not having the msys package installed as part of MinGW.  I was able to compile the bootloader after installing this as well as modifying the location of the arm gcc location in the makefile.  All in all I installed the following to build the bootloader in a windows environment:

ARM GNU tool chain:
https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads

NRF5x command tools:
https://www.nordicsemi.com/Software-and-tools/Development-Tools/nRF-Command-Line-Tools/Download#infotabs

MinGW (install make and msys)
https://osdn.net/projects/mingw/releases/
[image]

Git
https://gitforwindows.org/

Be sure to add locations of the programs to your environmental variables as needed.  You can test git, nrfjprog etc. in the command prompt to ensure that they are accessible.