adafruit/Adafruit_nRF52_Bootloader #109: [Bug] Not compiling for pca10059/feather_nrf52840_express

204maker, 2020-02-08:
**Describe the bug**
-Not compiling for pca10059/feather_nrf52840_express
-Missing nrfx_nvmc.h

**Set up (please complete the following information)**
  - pca10059, feather_nrf52840_express
  - Windows 10 Pro N on Asus Ultrabook i5

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'uf2-nRF52'
2. Compile bootloader for the board 'make BOARD=pca10059'
3. See error

**Expected behavior**
Bootloader file will be generated on the new folder '_build-pca10059'

**Screenshots**
See post on MakeCode forum.
[https://forum.makecode.com/t/makecode-for-nrf52840/1164/4]

hathach, 2020-02-08:
I cannot reproduce this issue, action ci build fine as well. Please try a clean git clone and post your error message here. 
https://github.com/adafruit/Adafruit_nRF52_Bootloader/runs/420479384?check_suite_focus=true

204maker, 2020-02-08:
It is weird. I did exactly what you suggested and it solved the issue. No errors during compilation. Thank you.

closed 2020-02-08 by 204maker