adafruit/Adafruit_nRF52_Bootloader #115: [Bug] Upgrading bootloader to 0.3.1 using jlink ended up in bootloader loop

akamensky, 2020-03-10:
**Describe the bug**
I decided to upgrade bootloader using my jlink from 0.3.0 to 0.3.1. After writing, I still "can" upload arduino sketches and other compiled binaries, however it is just stuck in bootloader (green led light up) and mass storage volume shows up repeatedly (every couple of minutes) named "FTHR840BOOT" with usual UF2 volume contents. Uploading any UF2 files to that volume results in no effect (no unmount and flash happens, it just stays there as mass volume).

**Set up (please complete the following information)**
  - feather_nrf52840_express
  - macos
  - ...

hathach, 2020-03-24:
try to upload and uf2 or an sketch via Serial once to settle the DFU setting. It should work afterward with jlink upload

akamensky, 2020-04-04:
I've recovered board somewhat, but it keeps going to bootloader after running valid sketch for some time (at random it looks).

hathach, 2020-04-05:
it is not a new bug, it is CRC checksum of bootloader, you should either flash the CRC16 of the application or disable the checking by writing 0x0001 (if I remembered correctly) to bootloader crc checking address (forget the exact address). Since you just erase and flash with jlink, simplest way is perfrom an DFU via serial or uf2, it iwll write the disable magice to crc checking.

closed 2020-04-05 by hathach

henrygab, 2020-04-12:
I had this same problem ... and I can confirm that installing a UF2 (I used [Circuit Python](https://circuitpython.org/board/pca10056/) as it was pre-generated) solved this bootloop for me just now.  What a relief!  I thought I had somehow damaged my DK, but it now boots sketches correctly again.  YAY!