adafruit/Adafruit_nRF52_Bootloader #116: [Bug] Cannot use LED_BUILTIN from Arduino on nRF52 Sense

colindgrant, 2020-03-11:
In order to make the the standard Arduino Blink sketch work on the Feather nRF52840 Sense, I had to replace `LED_BUILTIN` with the digital pin number `13` labeled on the board.

**Set up**
  - Feather nRF52840 Sense, reproduced with **0.2.9_s140_6.1.1** and **0.3.1_s140_6.1.1**
  - MacOS Mojave 10.14.6
  - Arduino 1.8.11 (latest), Adafruit nRF52 board definitions 0.18.5 (latest)
  - USB powered, no battery, no Featherwings, no other pin connections

**To Reproduce**
1. Connect Feather nRF52840 Sense via USB power/data
2. Open Arduino, select correct board and port (see below)
3. Put board into boot mode (verify BOOT volume mounted as USB drive)
4. Open Blink example, upload -> observe no blinking LED
5. Edit Blink example, s/LED_BUILTIN/13/g, upload again -> D13 red LED flashes

[image]

I validate that upload is successful two ways:
1. Neopixel goes from green to red to off
2. Arduino output shows success:
```
Uploading using selected port: /dev/cu.usbmodem14201
/Users/colin/Library/Arduino15/packages/adafruit/hardware/nrf52/0.18.5/tools/adafruit-nrfutil/macos/adafruit-nrfutil --verbose dfu serial -pkg /var/folders/s9/0skcmddj79ddc0z6s_40lll80000gn/T/arduino_build_36843/Blink.ino.zip -p /dev/cu.usbmodem14201 -b 115200 --singlebank 
Upgrading target on /dev/cu.usbmodem14201 with DFU package /private/var/folders/s9/0skcmddj79ddc0z6s_40lll80000gn/T/arduino_build_36843/Blink.ino.zip. Flow control is disabled, Single bank, Touch disabled
Opened serial port /dev/cu.usbmodem14201
Starting DFU upgrade of type 4, SoftDevice size: 0, bootloader size: 0, application size: 43532
Sending DFU start packet
Sending DFU init packet
Sending firmware file
########################################
########################################
######
Activating new firmware

Device programmed.
DFU upgrade took 3.368237018585205s
```

**Expected behavior**

The [Bluefruit nRF52 Feather Learning Guide](https://learn.adafruit.com/bluefruit-nrf52-feather-learning-guide/arduino-board-setup) section **4. Run a Test Sketch** uses the blink example with LED_BUILTIN, so I take that to imply that Adafruit expects the builtin keyword to work on this board. I know this is a fairly minor bug, but it is likely to confuse brand new users.

**Additional context**

The D13 red LED did not pulsate in BOOT mode when I first got the board. I came to this repo to get the latest firmware. After upgrading 0.2.9 to 0.3.1, the D13 red LED would fade on and off in BOOT mode as expected. So that is why I chose to file the issue in this repo. Please advise if it belongs in adafruit/Adafruit_nRF52_Arduino or somewhere else.

THANK YOU! Let me know if I can help further.

colindgrant, 2020-03-11:
I found out what was going on. 

It looks like the latest board definition available in Arduino (0.18.5) lacks the latest change(s) from Adafruit.

I needed this change:
https://github.com/adafruit/Adafruit_nRF52_Arduino/commit/4ae66084d86b7ae0c53add7233b63d080824071a

I edited my local file to match and now LED_BUILTIN works fine!
`~/Library/Arduino15/packages/adafruit/hardware/nrf52/0.18.5/variants/feather_nrf52840_sense.variant.h`

So I think this problem will resolve itself once the next version of adafruit/Adafruit_nRF52_Arduino is released. I'll test it out in a few days and close this issue.

colindgrant, 2020-03-11:
Or rather, just close it now. :)

closed 2020-03-11 by colindgrant

hathach, 2020-03-12:
thanks @colindgrant , it is indeed an typo for the sense variant. We will roll out the release in a day or two.