adafruit/Adafruit_nRF52_Bootloader #119: Need to Change BUTTON_FRESET Pin on Xenon Bootloader

Montvydas, 2020-04-13:
**Description**
In the bug that I raised within Adafruit Arduino repo (https://github.com/adafruit/Adafruit_nRF52_Arduino/issues/479#issue-598354165) I documented the problem that I was experiencing using Xenon board. To summarise everything, for Xenon (and other Particle boards) the Factory Reset Button (BUTTON_FRESET) is set to be A0. It took me quite some time to figure out why would my board always get stuck just after being flashed and the problem was that I used analog pin A0 to perform some measurements. Apparently if A0 is LOW (anything below ~1.5V) during Reset operation (which also includes flashing a new sketch), the device will perform a factory reset.

**Set up**
  - Xenon board with particle_xenon_bootloader-0.3.2_s140_6.1.1.hex bootloader
  - 2015 macbook pro 13" running Mojave
  - Serial debug log:
_BSP Library : 0.19.0
Bootloader : s140 6.1.1
Serial No : 3171D58826B5BECF_

**To Reproduce**
Check the linked issue I added, or simply connect pin A0 to GND and press reset button. Factory reset will be performed.

**Expected behavior**
As discussed in the linked issue, I believe a different pin should be set to perform such an operation as most of the users will intend to utilise pin A0 within their projects when ADC is needed (first analog pin, thus likely to be used?). At worst case scenario, this should at least be specified within Adafruit docs page for nRF52840, while also listing which pins are used to perform factory reset for each of the boards.

hathach, 2020-04-14:
indeed, it is an issue, I am not familiar with particle boards, do you have any suggestion for which pin we should change to. However, it could take time to release the bootloader, you could for now change the pin map for xenon and recompile the bootloader. It is rather straighforward.

PS: I am revising the bootloader, we may remove the FRST pin. Though for now, to keep it consistency, we need to map it to another pin.

Montvydas, 2020-04-14:
Well, my experience would be a week of playing with Xenon and nRF52840 Express and reading through the source code and schematics :D Based on that I got some mixed feelings when it comes to solving this. On one hand, Xenon is meant to be like a twin-brother of nRF52840 Express. However I noticed that Xenon corner pin NC (Not Connected) is in fact not connected to anything ([Xenon pinout](https://docs.particle.io/assets/images/xenon/xenon-pinout.png) and [schematics](https://github.com/particle-iot/xenon/blob/master/pdfs/xenon-v1.00-schematic.pdf) also checked the PCB file..), while nRF52840 Express defines it as both D2 and NC (Not Connected???). From the [schematics](https://learn.adafruit.com/assets/68545) this pin is connected to NFC2, so I believe the schematics was meant to say NFC rather than NC (mistyped?), but only guessing here. Now going to the point, within the nRF52840 Express bootloader this pin D2 is used as a BUTTON_FRESET, but 1-to-1 matching cannot be done on Xenon as nothing is connected there 😞Thus we either have to utilise another pin (currently A0) or remove this functionality completely. 

I believe, it does make 'some sense' keeping the BUTTON_FRESET. I do see it as a way of un-bricking an already deployed product in case it breaks. However, that depends on how slim are the chances of this happening in real world? I cannot answer that due to lack of experience of playing with it.. However, again looking at the Xenon pinout I do see a potentially less important pin D7/P1.12, which is a digital pin with PWM support, however it is not involved in any sort of comms, while the rest of the digital pins are and most of them also support PWM. Thus making D7 the 'least useful' pin 😄

For now I can simply use A1-A5 instead as there really isn't much of a different after knowing that A0 is forbidden and I only need one ADC pin anyway, but thanks for the suggestion!

hathach, 2020-04-14:
Ok, then we can keep this issue open for now. the FRST erase everything but bootloader + softdevice, which is not that useful. User can force DFU by double reset then flash an known good sketc, which is more convenient. FRST is more of legacy feature from feather nrf52832.

Montvydas, 2020-04-14:
Yeah, I did notice that on nrf52832 there was a dedicated test pad FRST. Thanks for telling about the double RESET click to put into DFU, really neat! I previously used the MODE button together with the RESET to achieve the same. Is there a way to put nrf52840 into OTA DFU without utilising FRST? If that's the case, I see no reason keeping the legacy functionality as it would not bring any value then.

hathach, 2020-04-15:
> Yeah, I did notice that on nrf52832 there was a dedicated test pad FRST. Thanks for telling about the double RESET click to put into DFU, really neat! I previously used the MODE button together with [...]

It is part of up coming release for bootloader. Though I am switching task to port USB for ESP32-S2 recently. Will get back to the soon. I was thinking to drop both DFU pin and FRST on 52840, and use only 1 GPIO to enter OTA. Normal bootloader should be enter by double reset. Though I still need to sort them out. Subscribe to the repo and release, I am sure will put that in the release note.