adafruit/Adafruit_nRF52_Bootloader #121: Simplify buttons usage

hathach, 2020-04-24:
**Is your feature request related to a problem? Please describe.**
Currently usage of bootloader is confusing and require 2 GPIOs which is quite a lot.

**Describe the solution you'd like**
FRST seems to be not needed since enter bootlaoder is easy enough, and user can just flash a known good firmware should their application corrupt. Therefore
- 832: Button1 to enter bootloader, button2 to enter OTA. No more combination of both
- 840: button to enter OTA. bootloader mode can be easily enter by double reset trick.

**Describe alternatives you've considered**
A better button-less for 840 is start both USB and OTA mode whenever enter bootloader. That would be optimal if we could that off.

@dhalbert @tannewt @PaintYourDragon for comments.

jpconstantineau, 2020-07-27:
For the 832 boards with no reset circuitry and off-board serial, would it be possible to enable double-reset to enter DFU mode?
Perhaps as an optional define option to set in the board.h file.  In previous releases, we could just press reset just in time when we were uploading to the board.  However, recently, the success rate of catching the 1000ms DFU window is pretty small.

hathach, 2020-07-27:
double reset is not possible with 832, Nordic did very hard job spending extra logic gates to make sure every bit of SRAM cleared upon reset.

jpconstantineau, 2020-07-27:
Humm. Do you mean that a pin reset clears the GPREGRET register?  That would indeed be problematic.

Could we leverage the RESETREAS register and detect if the reset reason was from the reset pin being triggered?
Depending on the reason, we could use different timing between DFU mode on bootup and DFU mode on pin reset.  Take for example, extend DFU_SERIAL_STARTUP_INTERVAL to 10 seconds on a pin reset and keep it short for other reasons. This would also allow to completely bypass DFU (or making it 10ms) on restart when waking up from sleep (making keyboards more responsive)

We may consider doing this as a non-feather/generic nrf52832 board so that it doesn't break the feathers of most people... (just in case I guess...)

hathach, 2020-07-28:
> Humm. Do you mean that a pin reset clears the GPREGRET register? That would indeed be problematic.

Humm. Do you mean that a pin reset clears the GPREGRET register? That would indeed be problematic. For reset reason, you can try it out to see what fit your need.

jpconstantineau, 2020-07-28:
I'll probably give it a try. However, it's just one more thing on my long
to-do list...

jpconstantineau, 2021-05-30:
I just configured a new 832 board that doesn't have reset circuitry and having a button to enter DFU is very useful before flashing from the Arduino IDE.  It's a simple as pressing "BUTTON_1", Pressing and releasing "Reset" followed by releasing "BUTTON_1".  This leaves the nrf52832 chip in DFU mode and the Arduino IDE is happy to flash a new sketch.

(I might be stating the obvious but this highlights the importance of "BUTTON_1" for boards that don't have serial normally installed or that don't have the reset circuitry.)

hathach, 2021-05-30:
832 will always have hold-button1 to DFU mode. Double reset is only possible with 840. Anyway, these should be configured via specific board header.

tannewt, 2021-08-02:
Double tap into USB + BLE would be awesome

markbirss, 2024-09-03:
@hathach 

Is the "clear user firmware and remain in bootloader mode" function defined in 

https://github.com/adafruit/Adafruit_nRF52_Bootloader/blob/eefbe6bda98b80cb296e2f1bb1c982d58fe4c12d/src/boards/waveshare_nrf52840_eval/board.h#L47

actually implemented and working ?