adafruit/Adafruit_nRF52_Bootloader #126: Decouple SoftDevice and self uf2

hathach, 2020-05-05:
**Is your feature request related to a problem? Please describe.**
Decouple softdevice along with adding self uf2. There is a duplicate issue of some sort, but I create this new one anyway.

**Describe the solution you'd like**
UF2 can be used to upgrade 
- Bootloader  (family ID = USB VID + PID) 
- Application (family ID = 0xADA52840)
- SoftDevice (either of above family ID)

Softdevice can be bundled together with either bootloader or application into a single UF2 file. CDC interface will left as it is as fail-safe since CDC can be used to revert to good-old version, should something bad happens.

**Check list**
- [ ] uf2 works with or without a SD on the flash
- [ ] uf2 works with application only
- [ ] uf2 works with softdevice only
- [ ] uf2 works with application + softdevice
- [ ] uf2 works with bootloader only
- [ ] uf2 works with bootloader + softdevice
- [ ] cdc works with application as it does now
- [ ] cdc works with bootloader + softdevice as it does now
- [ ] ota should work when SD is on the flash.

orrmany, 2020-05-06:
Hi @hathach , I am not sure if I understand the USB lingo, I am sorry.  
Basically this is my use case:

Suppose that I, as an Adafruit customer, previously purchased an Adafruit MCU with an nRF52 SoC onboard. 

Now, also suppose that I, as an Arduino developer, would like to be able replace the stock SoftDevice (SD) delivered with product with itd to a multiprotocol alternative, i fit exist. 
- For example, I would like to have the S140 SD of the nRF52840 Feather Express  replaced with S340. 
- Another example would be a nRF52832 customer who would like to replace S132 with S332. 
- etc.

I would like to retain all legacy Adafruit functionality with the alternative, multiprotocol SoftDevices, since the functionality of the alternative SD-s are extensions of that of the their lesser, BLE only counterparts. That is, I would like to use CircuitPython, DFU, Bluefruit, etc. with the multiprotocol SD-s.

Could you explain what of those checklist items of yours are applicable to my use case?

Thanks, Gábor

orrmany, 2020-05-06:
Oh, and I would like to retain the DFU capability over serial. For example, today I am using nRF52840 with a computer, where USB Mass Storage is blocked for administrative reasons. That is, UF2 (whatever UF2 means) is not an option for me.

hathach, 2020-05-06:
@orrmany please don't hijack this issue which is about decoupling Softdevice, you could ask S340 related issue on your own issue/pr.

orrmany, 2020-05-06:
Oh, I thought that checklist a questionnaire. Deleting comment here then

orrmany, 2020-05-06:
@hathach : how do you plan to handle the differences among the RAM allocation needs with/without softdevice? And the RAM allocations differences between alternative softdevices?

hathach, 2020-05-12:
> @hathach : how do you plan to handle the differences among the RAM allocation needs with/without softdevice? And the RAM allocations differences between alternative softdevices?

It is application specific and have nothing to do with the bootloader.

hathach, 2020-05-29:
implemented by #128

closed 2020-05-29 by hathach