adafruit/Adafruit_nRF52_Bootloader #130: Accept uf2 generated with family ID = VID/PID as board specific

hathach, 2020-05-15:
**Is your feature request related to a problem? Please describe.**

together with #129, accepting additional family as board specific (VID/PID) allow us to bundle multiple firmware of different boards into a single big uf2 (e.g a circuitpyhon uf2 file for all nRF52 boards ). Following is the original discussion

>FamilyID is meant as both board-id and MCU-id - for example, binaries generated from MakeCode Arcade would run on all NRF54840 boards provided they have the right bootloader settings (that define pinout, screen type, etc.). OTOH some other binaries may have pin configuration baked in, so they will only work on say 840 Feather. So it actually makes perfect sense for the bootloader to accept MCU-wide familyID and a specific boardID.
>
>Generally, this can thought of as a sub-divsion tree of devices and bootloader accepts everything on one path from root to leaf. In this case I think we have 3 levels (all UF2 file - familyID==0, NRF52840, specific board), but it could be more if needed (eg different revisions of a specific board).

_Originally posted by @mmoskal in https://github.com/adafruit/Adafruit_nRF52_Bootloader/pull/128_

**Describe the solution you'd like**
Bootloader accept family ID = `ada52840` as generic nrf binary and also family id = vid+pid as board specific one.

**Describe alternatives you've considered**
Application can also use pinconfig in CF2 of bootloader to write generic binary for all boards.

henrygab, 2020-07-26:
@hathach  -- See **_WIP_** PR #152.

It appears that allowing multiple family IDs, and basing the second allowed family ID on the USB VID/PID, may be relatively easy.

The above PR is a WIP, mostly because I've not built the bootloader (and am unsure if I can do so).  Thus, I created this to show that it *might* be a simple change.

hathach, 2020-07-27:
> @hathach -- See **_WIP_** PR #152.
> 
> It appears that allowing multiple family IDs, and basing the second allowed family ID on the USB VID/PID, may be relatively easy.
> 
> The above PR is still a WIP, mostly because I've not tested the bootloader (and am unsure if I can do so in any reasonable fashion). However, I created the PR to show that it _might_ be a simple change.

Thanks, It is much simpler than I thought. Was switched to other task by now :)

PR #152 (2020-08-03): Enable board-specific UF2 family IDs
**_NOTE: REQUIRES TESTING_**  (will edit if changes)

Fix #130.

For boards that do not define both USB VID and PID, no change in behavior.  Otherwise...

* Accept UF2 blocks that have the USB VID/PID as the family ID, **_in addition to_** the "legacy" family id of `0xada52840`.

* When exposing the current user application, exposes it using the USB VID/PID as the family ID.
closed 2020-08-03 by hathach