adafruit/Adafruit_nRF52_Bootloader #135: SPI Failure

ykqureshi, 2020-05-28:
**Hardware Setup**
- Adafruit nrf52840 Feather Express
- Bootloader version: "feather_nrf52840_express_bootloader-0.3.2_s140_6.1.1"
- IDE: VS code with platformio

**Problem Description**
The SPI communication doesn't work when the board is powered up with battery alone. Otherwise works fine when USB is plugged in. The problem is mainly related to VBUS. When I supply the VBUS with external 5V everything seems to work fine.

**To Reproduce**
Simply program the board with any SPI example and power up the board with battery or any 3.3v source alone. It won't communicate.
- PINs used: SCK, MOSI, MISO and CS = D10

**Expected behavior**
In my case, I have connected a fingerprint sensor FPC BM-Lite which is communicating through SPI. The sensor gets initialized well when USB is plugged as shown below:
[image]
But throws back communication error when USB is not plugged!!
[image]

**Data from logic analyzer**
When SPI is working fine I can easily see the communication
[image]

hathach, 2020-05-29:
are you SURE this is the bootloader's bug !!!??? Bootloader don't even enable SPI peripheral at all !!!!

ykqureshi, 2020-05-29:
> 
> 
> are you SURE this is the bootloader's bug !!!??? Bootloader don't even enable SPI peripheral at all !!!!

I am not 100% sure if this is directly related to bootloader but something wrong is happening in the very beginning. If I power up the board with USB then everything works fine otherwise  SPI doesn't work at all.

dhalbert, 2020-05-29:
Maybe try this with a vanilla Arduino SPI program as well, as opposed to platformIO.

ykqureshi, 2020-05-29:
I successfully tested the SPI with TFT module from Adafruit and it seems to fine. So I found a work around. I am initializing SPI TFT first and then initializing the fingerprint sensor. Problem solved. ;)
[image]

Thanks for your responses. 👍

closed 2020-05-29 by ykqureshi