adafruit/Adafruit_nRF52_Bootloader #138: Trying to use on nRF52840 DK (PCA10056) [Bug]

tusharvb19, 2020-06-13:
**Not working for me**

**Set up **
  - nRF52840 DK
  - Windows 10
  - No debug!!!

**To Reproduce**
Steps to reproduce the behavior:
1.  Connect DK and Flash bootloader using J-Link(Select board as nRF52840 DK, Tools - Burn bootloader while selecting Programmer as J-Link)
2. Now Flash the example bleuart using Upload using Programmer
3. No sign of working in nRF Connect mobile app

**Additional**
Also, Can you help to find the UART pins on nRF52840 DK?

JoeBakalor, 2020-06-13:
Did you program the SoftDevice or just the bootloader?

tusharvb19, 2020-06-13:
The Softdevice and Bootloader is included when we burn the bootloader, right?
I will confirm if both are installed using nRF connect pc-> Programmer!

tusharvb19, 2020-06-13:
Now when I flashed the code using Sketch--> Upload using Programmer

[image]

No sign...LED1 slowly blinks continuously!
And not showing anything in nRF Connect mobile app!

What are the pins for TX and RX for serial port to check for serial prints!

tusharvb19, 2020-06-15:
Anyone here to help me?

hathach, 2020-06-16:
you need to upload using Serial/Uf2 at least once or write magic number to bypass firmware integrity check. For more question, please post on the support forum.

closed 2020-06-16 by hathach

huddy987, 2020-07-16:
@hathach I am seeing the same issue. I tried uploading using UF2 as well, can you confirm that this should work?

1. make board=pca10056 all
2. make board=pca10056 flash
3. make board=pca10056 sd

Then upload S140 to the device using UF2. I am able to see the drive in USB cdc, but when I load an app, it does not jump to app (led stays pulsing for some reason)