adafruit/Adafruit_nRF52_Bootloader #146: Add unofficial support for ANT and Dual stack softdevices

huddy987, 2020-07-14:
Currently, if we want to build the bootloader with ANT or dual stack softdevices, we have to resort to hacking our own solution as @orrmany has done. From my understanding the current issue is that the ANT/Dual softdevices aren't allowed to be distributed, and so pre-patched bootloaders with the ANT/Dual softdevices can't be distributed either. My proposition is that we allow new targets in the makefiles to build bootloaders with these softdevices, but leave it up to the user to download the appropriate softdevice.

My idea is to change the makefile from:
make BOARD=XXXXX <Target>
to
make BOARD=XXXXX STACK=XXXXX <Target>
where stack is either "BLE" "ANT" or "BOTH"

When BLE is selected, we can handle building the same way we currently do.

For ANT, it's actually pretty simple. The S212 7.0.1 softdevice support nRF52832, nRF52810, nRF52833, and nRF52840. So we can use the S212 softdevice as a cover-all solution.

For dualstack, we would need to use S332 for nRF52832, and S340 for nRF52840. We could use 6.1.1, or update to 7.0.1 (doesn't really matter to me).

In any case, the main change would be to create linker scripts for each of these bootloaders, and compile switch in out BLE and ANT code for the single stack softdevices. If a user tries to build without having the ANT/dual stack softdevices, we can just prompt the user to download the stack from thisisant.com.

Of course, this would have to remain as unofficially supported because we can't actually distribute the proposed "new" bootloaders. Users of the "new" bootloaders would be on their own to interface with the softdevices ANT APIs in their apps. All this change would mean is that using ANT with adafruit's arduino library is a bit less tedious.

Thoughts? If this seems like a reasonable idea I can do the PR.

hathach, 2020-07-14:
SoftDevice is decoupled and not considered as part of bootloader anymore. You could just update it using uf2 file of any Sxxx you like. Feature is merged, just not released yet. ref https://github.com/adafruit/Adafruit_nRF52_Bootloader/pull/128

closed 2020-07-14 by hathach

huddy987, 2020-07-14:
There is some BLE code in the bootloader that won't work with the single ANT stack softdevice as well as sd_softdevice_init requires the ANT license parameter...it would be nice to at least have compile switch to compile in/out the required ANT changes

hathach, 2020-07-14:
can you be more specific to point out which function or line that doesn't work, and what the result of those running on your setup. We can discuss on how to address those issues.

huddy987, 2020-07-14:
I'll need to look into it deeper, but here's a couple preliminary issues that would need to be compile switched in/out

SD_softdevice_init requires the ANT license key parameters for ANT and DUAL stack softdevices:
https://github.com/adafruit/Adafruit_nRF52_Bootloader/blob/0b0677d0d7d9021736498d605535666404b87e3b/src/main.c#L330

SD_ble_enable is not a valid API call for ANT only softdevice. This will cause a hardfault:
https://github.com/adafruit/Adafruit_nRF52_Bootloader/blob/0b0677d0d7d9021736498d605535666404b87e3b/src/main.c#L367

You get the idea. Any BLE code in main.c will cause hardfaults on the ANT-only softdevice. For dualstack softdevices this shouldn't be a probably and can remain as is.

hathach, 2020-07-15:
Ok, for `sd_softdevice_enable` we just need to add an check to softdevice version S132 or S140 or ANT one to call the correct API. Therefore we still don't need to have it as compiler switch. Since I don't have access to any ANT stack, would you mind submitting a PR for detecting and initialize the stack probably ?

huddy987, 2020-07-15:
Oh yes true, we won’t need any compiler switch, we can just do the following checks for initializing the stack properly:

S3XX = ANT + BLE => Do ANT and BLE init
S2XX = ANT => Only init ANT
S1XX = BLE  => Only init BLE

I will take a look at this in the coming days

hathach, 2020-07-15:
the SD inited is for OTA, the current code is too specific for BLE. I would suggest in case of S2xx (ANT only), you should skip the OTA mode, since it does not likely to work. May need a bigger PR to support OTA over ANT.