adafruit/Adafruit_nRF52_Bootloader #150: [Bug] Unable to use NFCT

nitz, 2020-07-17:
**Describe the bug**
When using the UF2 bootloader, I'm unable to use the NFCT peripheral, despite the API calls looking okay.

**Set up (please complete the following information)**
- Using the [nRF52840 M.2 Dev Kit](https://github.com/makerdiary/nrf52840-m2-devkit) from makerdiary. For the board configuration, I'm basically using their fork's version [here](https://github.com/makerdiary/uf2-bootloader/tree/m2/src/boards/nrf52840_m2) but updated to have a pinconfig.c like the other boards. I've attached my current copy here [boards-nrf52840_m2.zip](https://github.com/adafruit/Adafruit_nRF52_Bootloader/files/4935804/boards-nrf52840_m2.zip). (Note the -U CFLAG in the mk, more on that in a bit.)
- Win10 Pro/Workstations 20H2 (19042.388)
- Not much serial log to show, as everything is pretty much generated from my output, but here's what it looks like anyways, once the application is running:
```
<debug> app: =====================================
<info> app: Application starting.
<debug> nfctagemu: Initializing Type 2 Tag NFCT via nrfx. (0x047D)
<info> NFCT: Initialized
<debug> nfctagemu: Payload set @0x200000EC
<info> nfctagemu: Tag enabled.
<info> NFCT: Start
<debug> app: Application waiting for events!
(the NFCT events never come...)
```

**To Reproduce**
Steps to reproduce the behavior:
1. Set up a board that you plan to use the NFCT on
2. Comment out the `CFLAGS += -DCONFIG_NFCT_PINS_AS_GPIOS` line from the main `Makefile`, so the system init doesn't try to skip the NFCT antenna pins. (I tried using `CFLAGS += -UCONFIG_NFCT_PINS_AS_GPIOS` in the `board.mk` file, but since it's evaluated before the line that *adds* the define, it never ends up being useful. Maybe we could move the CFLAGS defs above the included *.mk files?)
3. Flash the bootloader.
4. Use the bootloader to install an application with which to use the NFCT, using a phone reader software or other, observe it not working.
5. Using a non UF2 flash/bootloader, flash that same application. It works!

**Expected behavior**
I should be able to use the NFCT. I might just be missing something that I've not got set up.

**Screenshots**
N/A

**Additional context**
I went back through the commits to see why the `-DCONFIG_NFCT_PINS_AS_GPIOS` was added, and that definitely made sense. I do feel though that boards should be able to opt in to using the NFCT pins as NFCT pins via something in their `board.h`/`board.mk` Though it's clear that the define isn't my only problem :)

hathach, 2020-07-17:
I don't think we could help with this. Maybe someone that works more with NFC could, label as **help wanted**

nitz, 2020-07-17:
Sounds good! I plan to keep digging at it, as I'm sure it's almost gotta be just a register configured incorrectly or something. Will update with progress as I make any!

hathach, 2020-07-17:
NFC pin usage must be configured with UCIR, maybe the bootloader doesn't enable it by default. I kind of forgot, here is way to do it in application.
https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/libraries/Bluefruit52Lib/examples/Hardware/nfc_to_gpio/nfc_to_gpio.ino

If it is the case, we could definitely do it within board.h with an macro.

nitz, 2020-07-17:
Yeah, that's what the `CONFIG_NFCT_PINS_AS_GPIOS` seems to toggle, in `system_nrf52840.c` in NFRX.

```
    #if defined (CONFIG_NFCT_PINS_AS_GPIOS)
	#error If you can read this, you definitely don't have NFCT
 
        if ((NRF_UICR->NFCPINS & UICR_NFCPINS_PROTECT_Msk) == (UICR_NFCPINS_PROTECT_NFC << UICR_NFCPINS_PROTECT_Pos)){
            NRF_NVMC->CONFIG = NVMC_CONFIG_WEN_Wen << NVMC_CONFIG_WEN_Pos;
            while (NRF_NVMC->READY == NVMC_READY_READY_Busy){}
            NRF_UICR->NFCPINS &= ~UICR_NFCPINS_PROTECT_Msk;
            while (NRF_NVMC->READY == NVMC_READY_READY_Busy){}
            NRF_NVMC->CONFIG = NVMC_CONFIG_WEN_Ren << NVMC_CONFIG_WEN_Pos;
            while (NRF_NVMC->READY == NVMC_READY_READY_Busy){}
            NVIC_SystemReset();
        }   #endif
```
I stuck a `#error` in there to make sure the define was getting properly dropped when I commented it out of the Makefile, but I only just now stopped to wonder if `SystemInit()` is called by the bootloader. I guess I assumed it was, seemed "startuppy", but I haven't looked at it at all yet.

hathach, 2020-08-04:
could you tell us whether you figured out way to fix this ?

nitz, 2020-08-04:
Hey! I actually had stuck this on the backburner while I was focusing on my app, and was just using the default nRF5 MBR with no bootloader for now. But I've reached a "MVP" for my app, and am waiting on a PCB spin, so I will take a look at this today!

nitz, 2020-08-04:
Alrighty then! It looks like I got it working!

A comment in the `.ino` you linked actually may have uncovered what part of the issue was I was facing. It seems that once you write those UICR bits, unless you flash with a method that is going to re-flash them, they won't clear. (Despite my attempts to do so in software.) I think the *major* part of what happened was I had first flashed the bootloader with the "fresh out of the repo" build, which builds with `CONFIG_NFCT_PINS_AS_GPIOS` and `SOFTDEVICE_PRESENT` defined. 

With `CONFIG_NFCT_PINS_AS_GPIOS`, in system_nrf52840.c as I pointed out above, it does in fact set those bits. So even though I flashed over the bootloader with my builds later that took those defines out, it took a full chip erase and then re-flash of the bootloader without those defines present. So at the very least, it'll need to be optional for that flag to be defined (or rather, undefined?) by the board.mk

I just created a PR for the board I'm on currently (#155), and figure we can discuss on there how to best handle those symbol undefs!

nitz, 2020-09-14:
Closing this as #155 solved it!

closed 2020-09-14 by nitz