adafruit/Adafruit_nRF52_Bootloader #159: Getting error during compilation

ahtashambaig, 2020-08-12:
PS C:\Nordic\Ada_fruit bootloader\Adafruit_nRF52_Bootloader> make BOARD=pca10056 all
CC main.c
CC boards.c
CC pinconfig.c
CC nrfx_power.c
CC nrfx_nvmc.c
CC system_nrf52840.c
CC bootloader.c
CC bootloader_settings.c
CC bootloader_util.c
CC dfu_transport_serial.c
CC dfu_transport_ble.c
CC dfu_single_bank.c
CC ble_dfu.c
CC ble_dis.c
CC pstorage_raw.c
CC app_timer.c
CC app_scheduler.c
CC app_error.c
CC app_util_platform.c
CC crc16.c
CC hci_mem_pool.c
CC hci_slip.c
CC hci_transport.c
CC nrf_assert.c
CC msc_uf2.c
CC usb.c
CC usb_desc.c
CC ghostfat.c
CC dcd_nrf5x.c
CC tusb_fifo.c
CC usbd.c
CC usbd_control.c
CC cdc_device.c
CC msc_device.c
CC tusb.c
AS gcc_startup_nrf52840.S
LD pca10056_bootloader-.out
   text    data     bss     dec     hex filename
  31200    1572   22594   55366    d846 _build/build-pca10056/pca10056_bootloader-.out
Create pca10056_bootloader-.hex
Create pca10056_bootloader--nosd.hex
process_begin: CreateProcess(NULL, python3 tools/hexmerge.py --overlap=replace -o _build/build-pca10056/pca10056_bootloader--nosd.hex _build/build-pca10056/pca10056_bootloader-.hex lib/softdevice/mbr/hex/mbr_nrf52_2.4.1_mbr.hex, ...) failed.
make (e=2): The system cannot find the file specified.
make: *** [_build/build-pca10056/pca10056_bootloader--nosd.hex] Error 2
**Additional context**
Add any other context or screenshots about the feature request here.

hathach, 2020-08-12:
There is nothing to enhance

closed 2020-08-12 by hathach

ahtashambaig, 2020-08-13:
Hi, can you guide me with the procedure to compile this application ... I tried to compile for pca10056, but at first I got this as error:

src/main.c: In function 'main':
<command-line>: error: expected expression before '<<' token
src/main.c:183:34: note: in expansion of macro 'MK_BOOTLOADER_VERSION'
  183 |   BOOTLOADER_VERSION_REGISTER = (MK_BOOTLOADER_VERSION);
      |                                  ^~~~~~~~~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
make: *** [_build/build-feather_nrf52840_express/main.o] Error 1
PS C:\Nordic\Ada_fruit bootloader\Adafruit_nRF52_Bootloader>

I looked for the value of MK_BOOTLOADER_VERSION and directly put it there then I got the error as above

Do I need a specific file or location to put and compile this application

ahtashambaig, 2020-08-13:
Resolved issue was with python I didn't change python3 to python in Makefile

TG-Techie, 2020-08-31:
@ahtashambaig do you mean it needs python 2 as opposed to python 3? I'm having the same issue, would you mind giving a little more detail?

ahtashambaig, 2020-08-31:
No on Windows Python3 is installed with python named in path variables, in this repository you need to go to Makefile and replace python3 with python and save it
[image]
... it will be working if you compile it and get error at line 183 at this value MK_BOOTLOADER_VERSION, simply change the value with 0x00020A