adafruit/Adafruit_nRF52_Bootloader #163: [Bug] compilation error from bad `MK_BOOTLOADER_VERSION`

TG-Techie, 2020-08-31:
**Describe the bug**
When Compiling any board the build fails when compiling main.c, see the error below.

**Set up**
  - Board: feather_nrf52840_express and acustom board (testing express to verify build works correctly)
  - OS: A rather fresh install of macos 10.15.6 running python 3.8  
  - Serial debug log (enable IDE's Debug Mode Level to 1 or 2)  

**To Reproduce**
Steps to reproduce the behavior:
1. clone the repo on master
2. clone all submodules recursively 
3. install dependencies 
4. run ` make BOARD=feather_nrf52840_express all` or ` make BOARD=feather_nrf52840_express` with and without cleaning first
5. See error

**Expected behavior**
normal build cycle, run without major errors.

**The Error**
```
CC dfu_ble_svc.c
CC dfu_init.c
CC flash_nrf5x.c
CC main.c
src/main.c: In function 'main':
<command-line>:0:25: error: expected expression before '<<' token
src/main.c:184:34: note: in expansion of macro 'MK_BOOTLOADER_VERSION'
   BOOTLOADER_VERSION_REGISTER = (MK_BOOTLOADER_VERSION);
                                  ^~~~~~~~~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
make: *** [_build/build-feather_nrf52840_express/main.o] Error 1
```

**conjecture**
when searching the whole repo (including the submodules) there are only 3 mentions of `MK_BOOTLOADER_VERSION`:
- makefile line 290: `CFLAGS += -DMK_BOOTLOADER_VERSION='($(word 1,$(_VER)) << 16) + ($(word 2,$(_VER)) << 8) + $(word 3,$(_VER))'`
- main.c line 184: `  BOOTLOADER_VERSION_REGISTER = (MK_BOOTLOADER_VERSION);`
- an .emPower file ?

I think the issue is coming from line 290 in the make file. To me, it looks like this line is adding a preprocessor macro through the command line (see link 1) to `#define` the bootloader version number. To me the specified expressions looks fine, for either python or c, i can only assume there is some compounding error.

additionally, in issue 159 in this repo 'ahtashambaig' had a similar issue. however it seems that his issue stemed from a windows command name issue. 

any suggestions to fix it?

link 1: reaserach about `-D`: https://gcc.gnu.org/onlinedocs/gcc/Preprocessor-Options.html

TG-Techie, 2020-08-31:
I was able to hack a temporary fix in by hard coding MK_BOOTLOADER_VERSION as 0x00020A but that is only a short term solution

hathach, 2020-08-31:
try to run make with ``print-CFLAGS` and `print-_VER` to see their value in your setup, you probably missing some command. Here is my 

```
make BOARD=feather_nrf52840_express print-CFLAGS
CFLAGS = -DNRF52840_XXAA -DS140 -mthumb -mabi=aapcs -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -Os -ffunction-sections -fdata-sections -fno-builtin -fshort-enums -fstack-usage -fno-strict-aliasing -Wall -Wextra -Werror -Wfatal-errors -Werror-implicit-function-declaration -Wfloat-equal -Wundef -Wshadow -Wwrite-strings -Wsign-compare -Wmissing-format-attribute -Wno-endif-labels -Wunreachable-code -Wno-unused-parameter -Wno-expansion-to-defined -Wno-cast-function-type -D__HEAP_SIZE=0 -DCONFIG_GPIO_AS_PINRESET -DCONFIG_NFCT_PINS_AS_GPIOS -DSOFTDEVICE_PRESENT -DDFU_APP_DATA_RESERVED=7*4096 -DUF2_VERSION="0.3.2-145-g934d141-dirty lib/nrfx (v2.0.0) lib/tinyusb (0.6.0-272-g4e6aa0d8) lib/uf2 (heads/master)" -DBLEDIS_FW_VERSION="0.3.2-145-g934d141-dirty s140 6.1.1" -DMK_BOOTLOADER_VERSION=(0 << 16) + (3 << 8) + 2

make BOARD=feather_nrf52840_express print-_VER
_VER = 0 3 2
```

TG-Techie, 2020-08-31:
gladly. and ah!
```
me@my-laptop Adafruit_nRF52_Bootloader % make BOARD=logitech_G915TKL print-CFLAGS
CFLAGS = -DNRF52840_XXAA -DS140 -mthumb -mabi=aapcs -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -Os -ffunction-sections -fdata-section
s -fno-builtin -fshort-enums -fstack-usage -fno-strict-aliasing -Wall -Wextra -Werror -Wfatal-errors -Werror-implicit-function-declaration -W
float-equal -Wundef -Wshadow -Wwrite-strings -Wsign-compare -Wmissing-format-attribute -Wno-endif-labels -Wunreachable-code -Wno-unused-param
eter -Wno-expansion-to-defined -Wno-cast-function-type -D__HEAP_SIZE=0 -DCONFIG_GPIO_AS_PINRESET -DCONFIG_NFCT_PINS_AS_GPIOS -DSOFTDEVICE_PRE
SENT -DDFU_APP_DATA_RESERVED=7*4096 -DUF2_VERSION=" " -DBLEDIS_FW_VERSION=" s140 6.1.1" -DMK_BOOTLOADER_VERSION=( << 16) + ( << 8) +
me@my-laptop Adafruit_nRF52_Bootloader % make BOARD=logitech_G915TKL print-_VER
_VER =```

TG-Techie, 2020-08-31:
when adding `_GIT_VER_TG = "$(GIT_VERSION)"` above line 289 and printing it there is no output to the terminal.
copied:
```
me@my-laptop Adafruit_nRF52_Bootloader % make BOARD=logitech_G915TKL print-_GIT_VER_TG
_GIT_VER_TG =
```

hathach, 2020-08-31:
This is only environment setup not an repo bug. I think you should try to insert info/error to the makefile and try to run those git version to see if where is the command missing. I guess it is only either PATH, git or something.

TG-Techie, 2020-09-12:
still investigating

TG-Techie, 2020-09-12:
Okay, found the issue. and it is now building.

so the issue I had comes from the difference in gnu make versions. 
on lines 25 and 26of makefile the `!=` assignment operator is used, this is a feature added in 4.0 
`GIT_VERSION != git describe --dirty --always --tags`

I build my software on a mac (currently 10.15.6 ) which ships with GNU Make 3.81

the fix is to use the older version of this:
`GIT_VERSION = $(shell git describe --dirty --always --tags)`
those two lines are the only places in the file where that operator is used.

I'll make a pull request either adding the change to those lines or noting that make 4.0+ is required.
I'd lean towards the former but what do you think @hathach? @dhalbert?
was there a specific reason to use 4.0? 

a big thanks to @jepler and @jerryneedell  for helping diagnose this issue.

hathach, 2020-09-12:
please use the make v4, the shell command increase the compile time by 2 times.

hathach, 2020-09-12:
I tae that back Look back at that https://github.com/adafruit/Adafruit_nRF52_Bootloader/pull/147
. We can change it to use shell with `:=` t still compatible with v3 and compile time still the same. Please make a pr if you don't mind :)

TG-Techie, 2020-09-12:
Gladly!

BenRoe, 2020-10-10:
Had the same problem on OSX 10.14.6 and the changes from @TG-Techie helped.
Many thanks you saved my day.

PR #169 (2020-10-11): use shell cmd instead of assignment
fix #163
closed 2020-10-11 by hathach