adafruit/Adafruit_nRF52_Bootloader #164: Why boot drive timeout?

dhalbert, 2020-09-02:
With this UF2bootloader (not the Atmel one), there is a timeout after several minutes when the boot drive disappears. Currently this seems to be causing a Finder crash in some circumstances. See https://forums.adafruit.com/viewtopic.php?f=65&t=168987.

@hathach Is the timeout deliberate? What was the motivation? Thanks.

hathach, 2020-09-02:
Yes, the timeout is deliberate, it is there when we first started based on Nordic bootloader example. So I am not sure on the motivation either. We can remove it if you think it is better that way.

dhalbert, 2020-09-02:
Yes,thanks,  then, let's remove it, since it's causing actual crashes on vulnerable hosts (which shouldn't happen). I can't remember, is there a self-updater yet, or did that turn out to be impossible or extremely difficult?

kevinjwalters, 2020-09-03:
I timed an Adafruit CLUE (connected to a Windows host) at 300 (+/-1) seconds, btw. The posts mention 4 minutes. I assume this is just an estimation but it would be interesting if there is a difference here somewhere.

hathach, 2020-09-04:
oh, the bootloader is already updated that does not time out when uf2 is mounted, we just haven't released it yet. Though, we can remove it all for good as well.

PR #166 (2020-09-05): remove dfu timeout
fix #164
closed 2020-09-05 by dhalbert