adafruit/Adafruit_nRF52_Bootloader #170: [Bug] make all compilation error

Bhamoss, 2020-10-13:
**Describe the bug**
When compiling with `make BOARD=feather_nrf52840_express all`, compilation terminates with error as follows:
```
CC dfu_ble_svc.c
In file included from src/dfu_ble_svc.c:41:
lib/sdk11/components/libraries/bootloader_dfu/dfu_ble_svc.h:39:10: fatal error: nrf.h: No such file or directory
   39 | #include "nrf.h"
      |          ^~~~~~~
compilation terminated.
make: *** [Makefile:354: _build/build-feather_nrf52840_express/dfu_ble_svc.o] Error 1

```

**Set up (please complete the following information)**
  - Adafruit feather nRF52840 express
  - POP_OS 20.04
  - NA  

**To Reproduce**
Steps to reproduce the behaviour:
execute `make BOARD=feather_nrf52840_express all` in bash

**Expected behaviour**
A compilation with output files ready to flash.
I am looking for the version without SoftDevice.

**Additional contexts**
This is my first ever bug report, any faults in doing this are unintentional.

dhalbert, 2020-10-13:
Did you bring in the submodules? `git submodule update --init`

Bhamoss, 2020-10-13:
No, my apologies for wasting your time.

closed 2020-10-13 by Bhamoss