adafruit/Adafruit_nRF52_Bootloader #174: [Bug?] nRF52840 does not come out of reset after OTA DFU when connected to USB

biomurph, 2020-10-27:
**Describe the bug**
I've been banging on the nRF52840 and having some fun with the OTA bootloader and trying to DFU my module. I have an nRF52840 Express, and also a prototype board that I spun up myself. Both boards are showing the same behavior. 
Since I'm debugging, I have the board connected to USB for serial coms. I have used the Bluefruit Connect app on my cell as well as the nRF Toolbox DFU tool. Both show the same behavior. It all works well, connecting and such. Then finding the file(s) for the DFU tool to use, then starting bootloader, etc., and after a minute or so I get 'Success!'. However, the board is not up. I don't see it on my Bluetooth available connections, nor do I see it on the serial port. I have tried with an open serial port in Arduino (which is nice, since it is self-healing) and I've tried without any open serial port (just plugged in to USB). Only when I reset by pressing the reset pin do I get the board to 'wake up' and it runs the new sketch from the OTA DFU it just got.
This issue does not happen when I am running solely on battery power. If I'm not plugged to USB, the board (Express and my custom build) both jump right up and say 'hello' with a nice blinky LED.

Is this a bug? Or a feature?

I have **not** tried it with a power connection through the USB that is _not_ a computorb. 
The same behavior is seen when the board has a battery or doesn't have a battery.

**Set up**
Arduino Board Info:
BN: Adafruit Feather nRF52840 Express
VID: 239A
PID: 8029
SN: 3BF6E9CE7B9FB5C1
Mac OS
Arduino 1.8.13 to compile the sketch
My custom board was bootloaded with Arduino using my Segger tool. Super easy, by the way ;)

**To Reproduce**
I'm using the bleuart example sketch. The Bluefruit Connect or the nRF toolbox both have the same behavior after DFU.

**Expected behavior**
The board should come out of reset and communicate over the serial port, no?

ElreboCM, 2021-03-03:
I have related issue on the feather sense.

I am updating the bootloader as described [here](https://learn.adafruit.com/introducing-the-adafruit-nrf52840-feather/update-bootloader). 
That works fine but the feather does not come out of DFU. I have to manually copy UF2 CircuitPython image. 
Then the device comes out of DFU and runs the old sketch which is fine as I wanted to only update the bootloader.

ufanders, 2021-05-30:
Hey, I think I'm experiencing the same issue. I can only get my custom board to operate again with a power cycle, or by uploading my sketch via a J-Link, THEN uploading via DFU from the Arduino IDE.

Any movement on this issue?

SnappierSoap318, 2024-06-16:
bump

anything new?

nicogou, 2024-10-11:
Hi, I believe I am facing the same issue. Any news ?

beastoin, 2025-01-15:
https://github.com/adafruit/Adafruit_nRF52_Bootloader/issues/259