adafruit/Adafruit_nRF52_Bootloader #175: [Request] New Release

Nicell, 2020-11-02:
**Is your feature request related to a problem? Please describe.**
There's been a large amount of additions to the bootloader since the last release over 7 months ago.

**Describe the solution you'd like**
I'm proposing a new release is made, so that all of these new features and boards are readily available in a new release download.

**Describe alternatives you've considered**
I've been producing my own bootloader files locally, but it would be nice to have official versioned ones available from everyone.

hathach, 2020-11-10:
it needs a bit more testing, I will release it when I am back working with nrf52

hathach, 2021-01-19:
released as 0.4.0 https://github.com/adafruit/Adafruit_nRF52_Bootloader/releases/tag/0.4.0

closed 2021-01-19 by hathach