adafruit/Adafruit_nRF52_Bootloader #179: [Bug] Windows10 compilation issue

orrmany, 2020-12-04:
**Describe the bug**
Windows10 build failure

**Set up (please complete the following information)**
Adafruit nRF52840 Express
Windows 10
GNU Tools Arm Embedded 9 2019-q4-major

**To Reproduce**
Steps to reproduce the behavior:
1. git clone https://github.com/adafruit/Adafruit_nRF52_Bootloader
2. cd Adafruit_nRF52_Bootloader 
3. git submodule update --init --recursive
4. make BOARD=feather_nrf52840_express  all

**Expected behavior**
It shall compile

**Screenshots**
c:\local\Adafruit_nRF52_Bootloader>make BOARD=feather_nrf52840_express  all
CC dfu_ble_svc.c
CC dfu_init.c
CC flash_nrf5x.c
CC main.c
CC boards.c
CC pinconfig.c
CC nrfx_power.c
CC nrfx_nvmc.c
CC system_nrf52840.c
CC bootloader.c
CC bootloader_settings.c
CC bootloader_util.c
CC dfu_transport_serial.c
CC dfu_transport_ble.c
CC dfu_single_bank.c
CC ble_dfu.c
CC ble_dis.c
CC pstorage_raw.c
CC app_timer.c
CC app_scheduler.c
CC app_error.c
CC app_util_platform.c
CC crc16.c
CC hci_mem_pool.c
CC hci_slip.c
CC hci_transport.c
CC nrf_assert.c
CC msc_uf2.c
CC usb.c
CC usb_desc.c
CC ghostfat.c
CC dcd_nrf5x.c
CC tusb_fifo.c
CC usbd.c
CC usbd_control.c
CC cdc_device.c
CC msc_device.c
CC tusb.c
AS gcc_startup_nrf52840.S
LD feather_nrf52840_express_bootloader-0.3.2-184-g64ea299.out
   text    data     bss     dec     hex filename
  31348    1572   22618   55538    d8f2 _build/build-feather_nrf52840_express/feather_nrf52840_express_bootloader-0.3.2-184-g64ea299.out
Create feather_nrf52840_express_bootloader-0.3.2-184-g64ea299.hex
Create feather_nrf52840_express_bootloader-0.3.2-184-g64ea299-nosd.hex
process_begin: CreateProcess(NULL, python3 tools/hexmerge.py --overlap=replace -o _build/build-feather_nrf52840_express/feather_nrf52840_express_bootloader-0.3.2-184-g64ea299-nosd.hex _build/build-feather_nrf52840_express/feather_nrf52840_express_bootloader-0.3.2-184-g64ea299.hex lib/softdevice/mbr/hex/mbr_nrf52_2.4.1_mbr.hex, ...) failed.
make (e=2): The system cannot find the file specified.
make: *** [_build/build-feather_nrf52840_express/feather_nrf52840_express_bootloader-0.3.2-184-g64ea299-nosd.hex] Error 2

**Additional context**
Add any other context about the problem here.

orrmany, 2020-12-04:
@hathach: any kind advice and/or guidance what is the problem would be greatly appreciated. Thanks

orrmany, 2020-12-04:
Hmm, as I have written this report: could it be, that my Windows has python version 3, but it is not called "python3"?
'''
c:\local\Adafruit_nRF52_Bootloader>python --version
Python 3.9.0

c:\local\Adafruit_nRF52_Bootloader>python3
'python3' is not recognized as an internal or external command,
operable program or batch file.
'''

dhalbert, 2020-12-04:
That's right, depending on how python is installed on Windows, it gets different executable names, sometimes just `python`, sometimes `python` and `python3`. See this summary, written for another purpose: https://learn.adafruit.com/circuitpython-ble-libraries-on-any-computer/getting-python-on-your-host-computer#starting-python-3-on-windows-3075437-4.

I think you could add a `python3.exe` shortcut to `python.exe` to some directory that's in your `%PATH%`.

orrmany, 2020-12-04:
..and bingo, that has solved the problem. (I ran into several other minor problems with my adafruit-nrfutil installation and PATH settings, too.

Sorry for creating the issue, the bug was in my head, inbetween my monitor and chair....

closed 2020-12-04 by orrmany