adafruit/Adafruit_nRF52_Bootloader #184: Circuit PG Bluefruit - MakeCode with sound works only after first start

chatelao, 2021-01-06:
**Describe the bug**
A programm MakeCode with sound works only after first start

**Set up (please complete the following information)**
  - UF2 Bootloader 0.2.11-8-g2c13fd5-dirty lib/nrfx (v1.1.0-1-g096e770) lib/tinyusb (legacy-755-g55874813) s140 6.1.1
  - Model: Adafruit Circuit Playground nRF52840
  - Board-ID: nRF52840-CircuitPlayground-revD
  - Date: Jul 13 2019

**To Reproduce**
Steps to reproduce the behavior:
1. Go to MakeCode: https://makecode.com/_H8c6YtPJYMWf
2. Send programm to device
3. Push Button A/B (Should work)
4. Reset
5. Push Button A/B again (Will never work again)

**Expected behavior**
I didn't expect "the sounds of silence" ;-), but beeps again

hathach, 2021-01-08:
double reset to put CPB into bootloader then copy uf2 file to upgrade its firmware. I don't think CPB support the std way makecode reset board to bootloader.

chatelao, 2021-01-08:
The resetting works fine.

The sound doesn't work well after the first run. Unpower / repower doesn't work either.

hathach, 2021-01-08:
why do you think it is bootloader issue ? Can you try to upload with a simple blinky example with correct board selected from one of our arduino sketch to see if that works out.

chatelao, 2021-01-08:
Uploading works fine, it is only the audio behaviour with MakeCode. So if the bootloader has no audio code, your are right, this may be some weird startup / init behaviour, but within the Programm code, not the loader.

Btw: Updating the bootloader was very tedious (Arduino IDE & DFU setting required) compared to the SAMD21 boards, where just dropping a new UF2 file to update the bootloader was enough.

hathach, 2021-01-08:
closed since this is not bootloader issue

closed 2021-01-08 by hathach

chatelao, 2021-01-08:
Great to hear about the self-update.

=> Regarding the Audio: Maybe there is the PIN D11 (Power Saving on Audio) disable, is this touched during the booting?