adafruit/Adafruit_nRF52_Bootloader #187: [Bug] Updating the Bootloader to the latest 0.4.0 release

IeiuniumLux, 2021-02-01:
I'm trying to update my [Adafruit Feather nRF52840](https://www.adafruit.com/product/4516) Sense board with the latest [0.4.0 Bootloader release](https://github.com/adafruit/Adafruit_nRF52_Bootloader/releases/download/0.4.0/feather_nrf52840_sense_bootloader-0.4.0.zip) and the `adafruit-nrfutil` tool indicates that it was successfully upgraded, as shown below. However, the Arduino IDE still shows the previous version. 

[image]

[image]

I also tried to update it using the IDE (with Adafruit board and libraries updated) via the `Burn Booloader` menu option, but same result.  It completes successfully, but it continues showing the preview version 0.3.2.

hathach, 2021-02-02:
no this is not a bug, Arduino doesn't have ability to detect the bootloader, the menu is just a mean to show the bootloader that will be burned by the IDE. To check the actual version double reset to get device into uf2 mode and check its `INFO.txt` file. Seem like this cause some confusion, maybe we could remove this menu in the future.

closed 2021-02-02 by hathach

IeiuniumLux, 2021-02-02:
@hathach, As you pointed out, the `INFO_UF2.TXT` shows the latest Bootloader 0.4.0 version.  Thanks!

[image]

hathach, 2021-02-04:
I am glad that helps