adafruit/Adafruit_nRF52_Bootloader #212: I/O errors when copying UF2 file to Adafruit Feather nRF52840 Sense with Adafruit nRF52 Bootloader 0.6.0

koenvervloesem, 2021-07-22:
**Set up**
  - Adafruit Feather nRF52840 Sense with Adafruit nRF52 Bootloader 0.6.0
  - Host system: Ubuntu 20.04 LTS

**Describe the bug**

When I want to copy a UF2 file to the bootloader's mounted storage device, I get I/O errors and the file isn't copied.

**To Reproduce**

Steps to reproduce the behavior:
1. Enter bootloader mode by pushing the RESET button
2. Copy UF2 file for [CircuitPython](https://circuitpython.org/board/feather_bluefruit_sense/)
3. See error
4. CircuitPython doesn't run on the device

I have also tried other UF2 files, other USB cables and other USB ports of the host system.

**Dmesg output**

```
[211537.170865] usb 1-1: new full-speed USB device number 86 using xhci_hcd
[211537.325252] usb 1-1: New USB device found, idVendor=239a, idProduct=0087, bcdDevice= 1.00
[211537.325258] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[211537.325261] usb 1-1: Product: Feather nRF52840 Sense
[211537.325264] usb 1-1: Manufacturer: Adafruit Industries
[211537.325267] usb 1-1: SerialNumber: 468AB2181E24A3D8
[211537.327747] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[211537.328502] usb-storage 1-1:1.2: USB Mass Storage device detected
[211537.328810] scsi host3: usb-storage 1-1:1.2
[211538.335828] scsi host3: scsi scan: INQUIRY result too short (5), using 36
[211538.335838] scsi 3:0:0:0: Direct-Access     Adafruit nRF UF2          1.0  PQ: 0 ANSI: 2
[211538.336392] sd 3:0:0:0: Attached scsi generic sg2 type 0
[211538.337297] sd 3:0:0:0: [sdc] 65801 512-byte logical blocks: (33.7 MB/32.1 MiB)
[211538.337646] sd 3:0:0:0: [sdc] Write Protect is off
[211538.337650] sd 3:0:0:0: [sdc] Mode Sense: 03 00 00 00
[211538.337946] sd 3:0:0:0: [sdc] No Caching mode page found
[211538.337953] sd 3:0:0:0: [sdc] Assuming drive cache: write through
[211538.364478]  sdc:
[211538.367333] sd 3:0:0:0: [sdc] Attached SCSI removable disk
[211582.512099] usb 1-1: USB disconnect, device number 86
[211582.534654] blk_update_request: I/O error, dev sdc, sector 517 op 0x1:(WRITE) flags 0x0 phys_seg 1 prio class 0
[211582.534660] Buffer I/O error on dev sdc, logical block 517, lost async page write
[211582.735578] FAT-fs (sdc): unable to read boot sector to mark fs as dirty
```

hathach, 2021-07-26:
I have just tested, It seems like the update is complete regardless of the I/O error (due to bootloader reset itself). Which circuitpython version you are copying to. Try to use 6.x version, maybe there is some bug with 7.x one.

koenvervloesem, 2021-07-26:
I've tried this with adafruit-circuitpython-feather_bluefruit_sense-nl-6.3.0.uf2. I've tried some other UF2 files too, but I'll see whether I can find some other ones to install.

hathach, 2021-07-29:
@koenvervloesem with feather sense 6.3.0, does it work for you or it still has the issue.

koenvervloesem, 2021-07-29:
Unfortunately I'm still having the same issue, and I'm not sure what to try next.

How do you enter bootloader mode on your device? One push of the RESET button, right?

hathach, 2021-07-29:
yup, double reset to enter bootloader, can you post your info file contents here, this is mine. I have no issue with both cpy 6.3.0 and 7.0.0

```
UF2 Bootloader 0.6.0 lib/nrfx (v2.0.0) lib/tinyusb (0.10.1-41-gdf0cda2d) lib/uf2 (remotes/origin/configupdate-9-gadbb8c7)
Model: Adafruit Feather nRF52840 Sense
Board-ID: nRF52840-Feather-Sense
SoftDevice: S140 version 6.1.1
Date: Jun 19 2021
```

koenvervloesem, 2021-07-29:
My `INFO_UF2.TXT` says:

```
UF2 Bootloader 0.6.0 lib/nrfx (v2.0.0) lib/tinyusb (0.10.1-41-gdf0cda2d) lib/uf2 (remotes/origin/configupdate-9-gadbb8c7)
Model: Adafruit Feather nRF52840 Sense
Board-ID: nRF52840-Feather-Sense
SoftDevice: not found
Date: Jun 19 2021
```

This is the same as yours, except for the SoftDevice, which shouldn't make a difference because it's only the Bluetooth controller, right?

So installed the bootloader again, including the SoftDevice, and the info file now says:

```
UF2 Bootloader 0.6.0 lib/nrfx (v2.0.0) lib/tinyusb (0.10.1-41-gdf0cda2d) lib/uf2 (remotes/origin/configupdate-9-gadbb8c7)
Model: Adafruit Feather nRF52840 Sense
Board-ID: nRF52840-Feather-Sense
SoftDevice: S140 version 6.1.1
Date: Jun 19 2021
```

Which is exactly the same as yours now.

Then I tried again: I reset the Feather Sense, dragged the `adafruit-circuitpython-feather_bluefruit_sense-en_US-6.3.0` file to the drive, and again I get errors in `dmesg`, even more now:

```
[39927.902716] usb 1-3: USB disconnect, device number 33
[39927.917569] blk_update_request: I/O error, dev sdc, sector 11 op 0x1:(WRITE) flags 0x0 phys_seg 8 prio class 0
[39927.917583] buffer_io_error: 7 callbacks suppressed
[39927.917587] Buffer I/O error on dev sdc, logical block 11, lost async page write
[39927.917596] Buffer I/O error on dev sdc, logical block 12, lost async page write
[39927.917601] Buffer I/O error on dev sdc, logical block 13, lost async page write
[39927.917607] Buffer I/O error on dev sdc, logical block 14, lost async page write
[39927.917612] Buffer I/O error on dev sdc, logical block 15, lost async page write
[39927.917621] Buffer I/O error on dev sdc, logical block 16, lost async page write
[39927.917626] Buffer I/O error on dev sdc, logical block 17, lost async page write
[39927.917631] Buffer I/O error on dev sdc, logical block 18, lost async page write
[39927.917668] blk_update_request: I/O error, dev sdc, sector 269 op 0x1:(WRITE) flags 0x0 phys_seg 8 prio class 0
[39927.917673] Buffer I/O error on dev sdc, logical block 269, lost async page write
[39927.917679] Buffer I/O error on dev sdc, logical block 270, lost async page write
[39927.917740] blk_update_request: I/O error, dev sdc, sector 517 op 0x1:(WRITE) flags 0x0 phys_seg 1 prio class 0
[39928.021680] ------------[ cut here ]------------
[39928.021684] bdi-(unknown) not registered
[39928.021706] WARNING: CPU: 0 PID: 918552 at fs/fs-writeback.c:2326 __mark_inode_dirty+0x19e/0x3b0
...
[39928.021970] ---[ end trace fe4d73a70c3d4b82 ]---
[39928.024449] FAT-fs (sdc): Directory bread(block 517) failed
[39928.024458] FAT-fs (sdc): Directory bread(block 518) failed
[39928.024462] FAT-fs (sdc): Directory bread(block 519) failed
[39928.024472] FAT-fs (sdc): Directory bread(block 520) failed
[39928.024497] FAT-fs (sdc): Directory bread(block 517) failed
[39928.024501] FAT-fs (sdc): Directory bread(block 518) failed
[39928.024504] FAT-fs (sdc): Directory bread(block 519) failed
[39928.024508] FAT-fs (sdc): Directory bread(block 520) failed
[39928.126692] FAT-fs (sdc): unable to read boot sector to mark fs as dirty
```
And still the CIRCUITPY drive doesn't appear. I tried other cables too, still the same.

Do you know anything else I could try?

hathach, 2021-07-30:
are you using the EXACT feather sense board from adafruit or other board with similar schematics ?

koenvervloesem, 2021-07-30:
Here are some pictures of the board. Looks legit to me, and I bought it from a reputable web shop.

Front:

[image]

Back:

[image]

hathach, 2021-07-30:
> Here are some pictures of the board. Looks legit to me, and I bought it from a reputable web shop.

It look very genuine to me, and I have no doubt for that. I asked since normally Adafruit will ship with an older bootloader and SD pre-flashed. It is typically not easy to erase SD without jlink.

Back to the issu,e I have no particular idea why the upgrade failed for you, however, I guess it could be due to the qspi flash (hint by the I/0 error) and that could somehow hanged the circuitpython to run. Here is a few uf2 that compiled from arduino repo. Please try it out [bluefruit-sense-uf2.zip](https://github.com/adafruit/Adafruit_nRF52_Bootloader/files/6906405/bluefruit-sense-uf2.zip) for further analysis
1. bleuart : will act as bridge between cdc and bleuart (using bluefruit app). this app does not make use of external flash.
2. flash_info : will just read and print out the the flash device id and its size, simple to check qspi connection
3. flash_erase: will run a qspi flash chip erase
4. msc_external_flash: will present qspi flash as msc, you could try to format it within your PC, copy files over it, then power on/off to test its result.

Please try them out, hopefully it will provide more info.

koenvervloesem, 2021-07-30:
Yeah, the first thing I did before using the board was upgrading the bootloader, just a habit of me since I experienced some [problems with another board that were solved by upgrading the bootloader](https://koen.vervloesem.eu/blog/how-to-upgrade-the-adafruit-nrf52-bootloader/). No idea why the SoftDevice disappeared, though.

I tried flashing these four UF2 files now, but all of them still have the same problem: I see the same errors as previously in `dmesg` output.

At least the `bleuart` firmware should show itself as a BLE device, right? I don't see anything, also no `/dev/tty*` device appears or any other sign that Ubuntu discovers a USB device.

It doesn't seem like the code from the UF2 files is run, but that the errors happen during the flashing process.

So writing a bootloader UF2 file works (I could install the bootloader with SoftDevice previously, and I even successfully downgraded the bootloader to 0.5.1 to see whether this would make a difference), but writing any other UF2 file seems to fail.

hathach, 2021-08-03:
> Yeah, the first thing I did before using the board was upgrading the bootloader, just a habit of me since I experienced some [problems with another board that were solved by upgrading the [...]

Yeah, that is great habit. The SD missing is a bit odd though, but it is resolved now I guess

> I tried flashing these four UF2 files now, but all of them still have the same problem: I see the same errors as previously in `dmesg` output.
>
> At least the `bleuart` firmware should show itself as [...]
 
> It doesn't seem like the code from the UF2 files is run, but that the errors happen during the flashing process.
>
> So writing a bootloader UF2 file works (I could install the bootloader with [...]

hmm, seem like you have issue copying data from host to bootloader. One of the issue could be the host didn't cache the drive correctly, since the uf2 pretty discard other meta data and just extract the firmware payload to flash. If host system read back to verify the contents, it won't be able to response with the data that is written as host and could cause I/O error. 

Do you have any other PC to try out the updating just for hw sanity check ?

koenvervloesem, 2021-08-05:
Just tried this on another machine with the same Ubuntu 20.04 operating system, which shows the same error messages in `dmesg` output and results in the same behaviour. I also tried this on a Mac. I don't see anything relevant in `dmesg` output there, but the behaviour is the same: the code from the UF2 files doesn't run.

hathach, 2021-08-05:
this is really weird, you are able to flash using the CDC interface (adafruit-nrfutil), but seems to have issue using msc (uf2). Would you mind trying out the arduino IDE, It is relatively easy 
https://github.com/adafruit/Adafruit_nRF52_Arduino#recommended-adafruit-nrf52-bsp-via-the-arduino-board-manager

Please select the Feather Sense board, then try to compile a few built-in example e.g blinky and/or bleuart to see if it could be flashed via CDC interface and run. 

You may also try to upgrade bootloader using the Arduino IDE https://learn.adafruit.com/bluefruit-nrf52-feather-learning-guide/updating-the-bootloader .

hathach, 2021-11-30:
@koenvervloesem this issue is probably fixed by #214 caused when circuitpython flash size is > 512KB. Please update bootloader to at least 0.6.2 and try again. Thank you very much.

koenvervloesem, 2021-11-30:
I've just tested this with version 0.6.2 of the bootloader, and this works! Thanks for the reminder.

closed 2021-11-30 by koenvervloesem

hathach, 2021-11-30:
I am glad that works well, thank you for confirming it.