adafruit/Adafruit_nRF52_Bootloader #213: Unable to flash UF2 where flash is >512kb

tannewt, 2021-08-02:
**Set up**
Feather nRF52840 Express with 6.0.0 bootloader. CircuitPython e9369d50e16e557ccd1a98c06d6e36bb87b908b7 . UF2s for the opt build flash ok but DEBUG=1 builds do not.

**Describe the bug**

Only part of the >512k UF2 is flashed leaving the higher end of the flash incorrect. In CircuitPython this means the .data section is not correct and assertions in DEBUG=1 mode fail as a result.

**To Reproduce**

Steps to reproduce the behavior:
1. Enter the bootloader
2. Drag on
[firmware-debug.uf2.zip](https://github.com/adafruit/Adafruit_nRF52_Bootloader/files/6919834/firmware-debug.uf2.zip) after unzipping.
3. Red LED flashes quickly and board doesn't restart.
4. Manual restart leads to CP assertion failure.

Here is an opt build that works: [firmware-opt.uf2.zip](https://github.com/adafruit/Adafruit_nRF52_Bootloader/files/6919841/firmware-opt.uf2.zip)

Note that the broken UF2 is >1MB in size.

tannewt, 2021-08-02:
Two other things:
1. I wrote a script to compare the RAM version of .data to the flash version and they were the same. They were *not* the same as the .data version from the .elf file.
2. The debug build of CircuitPython works when I flash it via pyocd instead of with the bootloader.

dhalbert, 2021-08-02:
Thank you for figuring this out!

hathach, 2021-08-03:
yeah, I could reproduce the issue, troubleshooting on the way, will post more update when available.

hathach, 2021-08-04:
```
Write addr = 0x000ACD00, block = 2142 (2142 of 2191)
Write addr = 0x000ACE00, block = 2143 (2143 of 2191)
Write addr = 0x000ACF00, block = 2144 (2144 of 2191)
```

I found the issue, the bootloader is limited its size to 512KB since it reserved the rest for internal file system ( for all boards), since the very first implementation.
- For Arduino, all boards is served with 28KB for storing bonding information. 
- For Circuitpython, there seems to be 3 section FLASH_BLE_CONFIG (32KB), FLASH_NVM (8KB), FLASH_FATFS (can be 0 on qspi flash dev).

For that I think we will increase reserved size to 40KB. The overflow to fatfs should be checked by circuitpyhon  linker, just try DEBUG=1 with boards internal flash, it overflows and won't linked which is good.

[image]

PR #214 (2021-08-04): Fix flash with more than 512kb
fix #213 

```
Write addr = 0x000ACD00, block = 2142 (2142 of 2191)
Write addr = 0x000ACE00, block = 2143 (2143 of 2191)
Write addr = 0x000ACF00, block = 2144 (2144 of 2191)
```

I found the issue, the bootloader is limited its size to 512KB since it reserved the rest for internal file system ( for all boards), since the very first implementation of the bootloader.
- For Arduino, all boards is served with 28KB for storing bonding information. 
- For Circuitpython, there seems to be 3 section FLASH_BLE_CONFIG (32KB), FLASH_NVM (8KB), FLASH_FATFS (can be 0 on qspi  [... cut]
closed 2021-08-04 by hathach