adafruit/Adafruit_nRF52_Bootloader #220: Xenon bootloader update from 0.5.0 to 0.6.1 not working?

ScruffR, 2021-09-07:
**Set up**
  - Particle Xenon
  - Windows 10 (21H1 Build 19043-1202)
  - SW on the device (bootloader mode)
```
UF2 Bootloader 0.5.0 lib/nrfx (v2.0.0) lib/tinyusb (0.9.0-22-g7cdeed54) lib/uf2 (remotes/origin/configupdate-9-gadbb8c7)
Model: Particle Xenon
Board-ID: nRF52840-Xenon-v1
SoftDevice: S140 version 6.1.1
Date: Apr  4 2021
```

**Describe the bug**

I did expect that I can replace the current bootloader by putting the device in bootloader mode and copying the `update-particle_xenon_bootloader-0.6.1_nosd.uf2` file to the device. 
However when the device comes back ist still reports bootloader 0.5.0

Is it not supposed to work as UF2 upload or is that file incompatible with the Xenon?

hathach, 2021-09-07:
it should work as other boards, but I don't have a xenon to test with.

ScruffR, 2021-09-07:
@hathach , thanks for the reply :+1:

So you also think that my process was correct and should work?

 - plug in device
 - put device into bootloader mode (i.e. press/hold MODE button and tap RESET)
 - wait for device's onboard LED to turn green and expose BOOT drive (i.e. XENONBOOT) on host computer
 -  copy .UF2 file to BOOT drive
 - wait for device to restart

hathach, 2021-09-08:
Yes, your procedure is correct

closed 2021-11-29 by hathach