adafruit/Adafruit_nRF52_Bootloader #224: error: "NRF_REGULATORS_HAS_POFCON" is not defined

ada19911201, 2021-09-19:
### Operating System

Windows 10

### INFO_UF2.TXT

feather_nrf52840_express

### What happened ?

In file included from lib/nrfx/drivers/include/nrfx_power.h:40:0,
                 from lib/nrfx/drivers/src/nrfx_power.c:38:
lib/nrfx/drivers/include/nrfx_power_compat.h:49:5: error: "NRF_REGULATORS_HAS_POFCON" is not defined, evaluates to 0 [-Werror=undef]
 #if NRF_REGULATORS_HAS_POFCON
     ^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from lib/nrfx/drivers/include/nrfx_power.h:40:0,
                 from src/main.c:43:
lib/nrfx/drivers/include/nrfx_power_compat.h:49:5: error: "NRF_REGULATORS_HAS_POFCON" is not defined, evaluates to 0 [-Werror=undef]
 #if NRF_REGULATORS_HAS_POFCON
     ^~~~~~~~~~~~~~~~~~~~~~~~~
ccoommppiillaattiioonn  tteerrmmiinnaatteedd  dduuee  ttoo  --WWffaattaall--eerrrroorrss..

### How to reproduce ?

I use make  BOARD=feather_nrf52840_express all

jerryneedell, 2021-09-19:
hmm - it boils OK for me.
Did you do update the submodules aft cloning the repository:
```
git clone https://github.com/adafruit/Adafruit_nRF52_Bootloader
cd Adafruit_nRF52_Bootloader
git submodule update --init
```  per the README.md

ada19911201, 2021-09-20:
I'm using GCC version 7.3.1, I'll update to 10.2 to try it out

closed 2021-09-20 by ada19911201