adafruit/Adafruit_nRF52_Bootloader #229: Compatibility with Raytac's new MDBT53-1M series of chip (nRF5340)

dsteinman, 2021-11-03:
MDBT50Q is out of stock EVERYWHERE.  It's nuts out there.  I am considering a change to the new MDBT53.

https://www.raytac.com/product/ins.php?index_id=121

https://raytaccorp.blogspot.com/2020/12/nrf5340-module-mdbt53-series-by-raytac.html

I'm guessing this nRF52 bootloader isn't going to work on these new nRF5340 chips?

hathach, 2021-11-03:
tinyusb is compatible with 5340, so it is just a matter of porting works. I will be happy to review if you make an PR for this.

dsteinman, 2021-11-03:
Unfortunately I don't think I have the skillset to do this myself, I wouldn't even know where to start.

hathach, 2021-11-04:
I don't the binary is compatible with 5340. I could be wrong, you may want to clarify this on Nordic devzone. The work is probably straight forward, unfortunately I don't have any 5340 devkit to help.

closed 2021-11-05 by hathach