adafruit/Adafruit_nRF52_Bootloader #244: compile error with DEBUG=1

bigbearishappy, 2021-11-30:
### Operating System

Linux

### INFO_UF2.TXT

UF2 Bootloader 0.6.2-12-g459adc9-dirty lib/nrfx (v2.0.0) lib/tinyusb (0.10.1-293-gaf8e5a90) lib/uf2 (remotes/origin/configupdate-9-gadbb8c7)
Model: Adafruit Feather nRF52840 Express
Board-ID: nRF52840-Feather-revD
Date: Nov 29 2021
SoftDevice: S140 7.3.0

### What happened ?

I can compile the bootloader successfully with the following command:
make BOARD=feather_nrf52840_express all
But,when I want to enable the printf in the booloader with DEBUG=1 in compile command:
make BOARD=feather_nrf52840_express DEBUG=1 all
I got the following error:
`LD feather_nrf52840_express_bootloader-0.6.2-12-g459adc9-dirty.out
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: _build/build-feather_nrf52840_express/feather_nrf52840_express_bootloader-0.6.2-12-g459adc9-dirty.out sectio                        n `.text' will not fit in region `FLASH'
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: region FLASH overflowed with .data and user data
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: section .bootloaderConfig LMA [00000000000fd800,00000000000fd857] overlaps section .text LMA [00000000000f40                        00,00000000000fd947]
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: region `FLASH' overflowed by 336 bytes
collect2: error: ld returned 1 exit status
make: *** [Makefile:393: _build/build-feather_nrf52840_express/feather_nrf52840_express_bootloader-0.6.2-12-g459adc9-dirty.out] Error 1
`

### How to reproduce ?

make BOARD=feather_nrf52840_express DEBUG=1 all

### Debug Log

CC dfu_ble_svc.c
CC dfu_init.c
CC flash_nrf5x.c
CC main.c
CC boards.c
CC nrfx_power.c
CC nrfx_nvmc.c
CC system_nrf52840.c
CC bootloader.c
CC bootloader_settings.c
CC bootloader_util.c
CC dfu_transport_serial.c
CC dfu_transport_ble.c
CC dfu_single_bank.c
CC ble_dfu.c
CC ble_dis.c
CC pstorage_raw.c
CC app_timer.c
CC app_scheduler.c
CC app_error.c
CC app_util_platform.c
CC crc16.c
CC hci_mem_pool.c
CC hci_slip.c
CC hci_transport.c
CC nrf_assert.c
CC pinconfig.c
CC msc_uf2.c
CC usb_desc.c
CC usb.c
CC ghostfat.c
CC dcd_nrf5x.c
CC tusb_fifo.c
CC usbd.c
CC usbd_control.c
CC cdc_device.c
CC msc_device.c
CC tusb.c
CC SEGGER_RTT.c
AS gcc_startup_nrf52840.S
LD feather_nrf52840_express_bootloader-0.6.2-12-g459adc9-dirty.out
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: _build/build-feather_nrf52840_express/feather_nrf52840_express_bootloader-0.6.2-12-g459adc9-dirty.out sectio                        n `.text' will not fit in region `FLASH'
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: region FLASH overflowed with .data and user data
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: section .bootloaderConfig LMA [00000000000fd800,00000000000fd857] overlaps section .text LMA [00000000000f40                        00,00000000000fd947]
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: region `FLASH' overflowed by 336 bytes
collect2: error: ld returned 1 exit status
make: *** [Makefile:393: _build/build-feather_nrf52840_express/feather_nrf52840_express_bootloader-0.6.2-12-g459adc9-dirty.out] Error 1

hathach, 2021-11-30:
yes, the debug code is too large to fit into bootloader flash region.

bigbearishappy, 2021-11-30:
> yes, the debug code is too large to fit into bootloader flash region.

Thank you for your reply.
it seems that I need to disable some function in bootloader to make the printf work fine.
Is there any function can be disabled optionally?

hathach, 2022-10-25:
fixed by #280

closed 2022-10-25 by hathach