adafruit/Adafruit_nRF52_Bootloader #269: DFU OTA upgrade problem

AndreasLamparter, 2022-07-05:
### Operating System

Windows 10

### INFO_UF2.TXT

UF2 Bootloader 0.6.4 lib/nrfx (v2.0.0) lib/tinyusb (0.12.0-145-g9775e7691) lib/uf2 (remotes/origin/configupdate-9-gadbb8c7)
Model: Adafruit Feather nRF52840 Sense
Board-ID: nRF52840-Feather-Sense
Date: May 17 2022
SoftDevice: S140 6.1.1

### What happened ?

I want to update my application my adafruit feather sense over DFU OTA. I use the firmware.zip created by PlatformIO and ArduinoIDE.
I try nRCConnect and the DFU Upgrade tool. On both tools the upgrade stuck after firmware.zip upload is complete.

I use the dfu_ota.ino sketch to test.
```
#include <Arduino.h>

void setup() 
{
  enterOTADfu();
}

void loop() 
{
}
```

The PlatformIO cmd to create the firmeware.zip

`tool-adafruit-nrfutil\adafruit-nrfutil.py dfu genpkg --dev-type 0x0052 --sd-req 0x00B6 --application .\firmware.hex .\firmware.zip`

I checked the --sd-req 0x00B6 parameter, it should match with the SoftDevice S140 6.1.1

### How to reproduce ?

1. Connect the nRF Connect app with the device.
2. Start DFU with select the firmeware.zip
3. The update stuck after uploading the firmeware.

### Debug Log

nRF Connect, 2022-07-05
AdaDFU (EF:DD:77:9B:F7:45)
V 17:57:37.388 Connecting to EF:DD:77:9B:F7:45...
D 17:57:37.388 gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferred PHY = LE 1M)
D 17:57:37.677 [Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I 17:57:37.677 Connected to EF:DD:77:9B:F7:45
D 17:57:37.690 [Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
V 17:57:37.699 Discovering services...
D 17:57:37.700 gatt.discoverServices()
I 17:57:38.017 Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
D 17:57:38.399 [Callback] Services discovered with status: 0
I 17:57:38.399 Services discovered
V 17:57:38.422 Generic Access (0x1800)
- Device Name [R W] (0x2A00)
- Appearance [R] (0x2A01)
- Peripheral Preferred Connection Parameters [R] (0x2A04)
- Central Address Resolution [R] (0x2AA6)
Generic Attribute (0x1801)
- Service Changed [I] (0x2A05)
   Client Characteristic Configuration (0x2902)
Device Firmware Update Service (00001530-1212-efde-1523-785feabcd123)
- DFU Packet [WNR] (00001532-1212-efde-1523-785feabcd123)
- DFU Control Point [N W] (00001531-1212-efde-1523-785feabcd123)
   Client Characteristic Configuration (0x2902)
- DFU Version [R] (00001534-1212-efde-1523-785feabcd123)
Device Information (0x180A)
- Manufacturer Name String [R] (0x2A29)
- Model Number String [R] (0x2A24)
- Firmware Revision String [R] (0x2A26)
D 17:57:38.422 gatt.setCharacteristicNotification(00002a05-0000-1000-8000-00805f9b34fb, true)
I 17:57:38.443 Connection parameters updated (interval: 48.75ms, latency: 0, timeout: 5000ms)
V 17:58:23.779 [DFU] DFU service started
V 17:58:23.779 [DFU] Opening file...
I 17:58:23.813 [DFU] Firmware file opened successfully
V 17:58:23.813 [DFU] Connecting to DFU target...
D 17:58:23.813 [DFU] gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferredPhy = LE_1M | LE_2M)
I 17:58:23.829 [DFU] Connected to EF:DD:77:9B:F7:45
V 17:58:23.829 [DFU] Discovering services...
D 17:58:23.829 [DFU] gatt.discoverServices()
I 17:58:23.829 [DFU] Services discovered
D 17:58:23.846 [DFU] wait(1000)
V 17:58:24.888 [DFU] Reading DFU version number...
D 17:58:24.888 [DFU] gatt.readCharacteristic(00001534-1212-efde-1523-785feabcd123)
I 17:58:24.930 [DFU] Read Response received from 00001534-1212-efde-1523-785feabcd123, value (0x): 08-00
A 17:58:24.931 [DFU] Version number read: 0.8
V 17:58:24.931 [DFU] Requesting new MTU...
D 17:58:24.931 [DFU] gatt.requestMtu(517)
I 17:58:25.035 [DFU] MTU changed to: 247
D 17:58:25.035 [DFU] wait(1000)
V 17:58:26.019 [DFU] Enabling notifications for 00001531-1212-efde-1523-785feabcd123
D 17:58:26.019 [DFU] gatt.setCharacteristicNotification(00001531-1212-efde-1523-785feabcd123, true)
D 17:58:26.019 [DFU] gatt.writeDescriptor(00002902-0000-1000-8000-00805f9b34fb, value=0x01-00)
I 17:58:26.102 [DFU] Data written to descr.00001531-1212-efde-1523-785feabcd123, value (0x): 01-00
V 17:58:26.102 [DFU] Notifications enabled for 00001531-1212-efde-1523-785feabcd123
A 17:58:26.102 [DFU] Notifications enabled
D 17:58:26.102 [DFU] wait(1000)
V 17:58:27.108 [DFU] Writing to characteristic 00001531-1212-efde-1523-785feabcd123
D 17:58:27.108 [DFU] gatt.writeCharacteristic(00001531-1212-efde-1523-785feabcd123)
I 17:58:27.174 [DFU] Data written to 00001531-1212-efde-1523-785feabcd123, value (0x): 01-04
A 17:58:27.174 [DFU] DFU Start sent (Op Code = 1, Upload Mode = 4)
V 17:58:27.174 [DFU] Writing to characteristic 00001532-1212-efde-1523-785feabcd123
D 17:58:27.174 [DFU] gatt.writeCharacteristic(00001532-1212-efde-1523-785feabcd123)
I 17:58:27.174 [DFU] Data written to 00001532-1212-efde-1523-785feabcd123, value (0x): 00-00-00-00-00-00-00-00-F0-75-02-00
A 17:58:27.174 [DFU] Firmware image size sent (0b, 0b, 161264b)
I 17:58:35.075 [DFU] Notification received from 00001531-1212-efde-1523-785feabcd123, value (0x): 10-01-01
A 17:58:35.075 [DFU] Response received (Op Code = 1 Status = 1)
A 17:58:35.075 [DFU] Writing Initialize DFU Parameters...
V 17:58:35.075 [DFU] Writing to characteristic 00001531-1212-efde-1523-785feabcd123
D 17:58:35.075 [DFU] gatt.writeCharacteristic(00001531-1212-efde-1523-785feabcd123)
I 17:58:35.157 [DFU] Data written to 00001531-1212-efde-1523-785feabcd123, value (0x): 02-00
V 17:58:35.157 [DFU] Writing to characteristic 00001532-1212-efde-1523-785feabcd123
D 17:58:35.157 [DFU] gatt.writeCharacteristic(00001532-1212-efde-1523-785feabcd123)
I 17:58:35.157 [DFU] Data written to 00001532-1212-efde-1523-785feabcd123, value (0x): 52-00-FF-FF-FF-FF-FF-FF-01-00-B6-00-A0-29
V 17:58:35.157 [DFU] Writing to characteristic 00001531-1212-efde-1523-785feabcd123
D 17:58:35.157 [DFU] gatt.writeCharacteristic(00001531-1212-efde-1523-785feabcd123)
I 17:58:35.308 [DFU] Notification received from 00001531-1212-efde-1523-785feabcd123, value (0x): 10-02-01
I 17:58:35.308 [DFU] Data written to 00001531-1212-efde-1523-785feabcd123, value (0x): 10-02-01
A 17:58:35.308 [DFU] Initialize DFU Parameters completed
A 17:58:35.308 [DFU] Response received (Op Code = 2, Status = 1)
V 17:58:35.308 [DFU] Writing to characteristic 00001531-1212-efde-1523-785feabcd123
D 17:58:35.308 [DFU] gatt.writeCharacteristic(00001531-1212-efde-1523-785feabcd123)
I 17:58:35.409 [DFU] Data written to 00001531-1212-efde-1523-785feabcd123, value (0x): 03
A 17:58:35.409 [DFU] Receive Firmware Image request sent
A 17:58:35.410 [DFU] Uploading firmware...
V 17:58:35.410 [DFU] Sending firmware to characteristic 00001532-1212-efde-1523-785feabcd123...

clonedesign, 2022-07-29:
Hi, is there any update about this issue?

baconcheese113, 2022-08-27:
Getting the same error from the `dfu_ota.ino` example sketch. 

I've tried using the .zip and .hex output from the Arduino IDE compilation at 
`C:\Users\{user}\AppData\Local\Temp\arduino-sketch-3D6AB27382EB0CF2C59607F00AD8FAC1` 
As well as the .zip and .hex output from platformIO compilation at 
`C:\Users\{user}\Documents\PlatformIO\Projects\{project}\.pio\build\{env}`

But I get the same error with both.

With **Bluefruit Connect** the app is able to connect to my advertising device fine, it can read the GATT server correctly, but when trying to perform an OTA DFU it brings up a `Starting DFU Update...` window and freezes.

With **nRF Connect** the app is able to connect and read the GATT server, and shows the DFU option in the app bar, but then when performing the OTA DFU it freezes after transferring the whole file. It is able to get the board into bootloader mode, but gets stuck there. Future resets continue back to the bootloader and it has to be re-flashed to run the sketch again

aqibnasim, 2023-01-02:
I have also observed the same issue. Apparently it seems that its due to latest Android versions. BLE DFU seems to work fine with nrf application deployed with older versions of Android:
https://apkpure.com/nrf-toolbox-for-bluetooth-le/no.nordicsemi.android.nrftoolbox/download/78-APK

Montvydas, 2023-02-21:
This seems to be the same problem as #287. The problem is created due to the changes to the Nordic secure bootloader, which now supports MTU sizes larger than 23. Before that it was always forcing MTU to be 23 regardless of the bootloader, but now it allows the bootloader to decide on the MTU size itself. The older unsecure booloader, such as the Adafruit one doesn't support MTU sizes other than 23, which is a problem since it requests MTU size of 247, which is incompatible and thus it breaks. There is a merge request #293, which should fix this issue.