adafruit/Adafruit_nRF52_Bootloader #277: PCA10056: Part of the HEX regions are out of the device memory size

lonelycorn, 2022-09-16:
### Operating System

Windows 11

### INFO_UF2.TXT

N/A

### What happened ?

I downloaded [pca10056_bootloader-0.7.0_s140_6.1.1.hex](https://github.com/adafruit/Adafruit_nRF52_Bootloader/releases/download/0.7.0/pca10056_bootloader-0.7.0_s140_6.1.1.hex) from the 0.7.0 release, and used Programmer v3.0.3 from [nRF Connect](https://www.nordicsemi.com/Products/Development-tools/nrf-connect-for-desktop) to flash an [nRF52840-DK](https://www.nordicsemi.com/Products/Development-hardware/nrf52840-dk). 

The programmer showed the following warning (see screenshot)
```
Part of the HEX regions are out of the device memory size, but you can still proceed write operation
```

The device seems to be working fine-ish after the flash. At least for a couple of days I was able to flash Arduino sketches using the Arduino IDE via the adafruit package.

But then the GPIO became dysfunctional. For example, the LED does not blink with Arduino "Blink" example sketch. Using a scope, I found that P0.13 (driving LED1) outputs 1.7V.

When I flashed the [Zephyr "Blinky" sample](https://docs.zephyrproject.org/2.7.0/samples/basic/blinky/README.html), the LED worked fine.

Flashing back to the Arduino bootloader and the "Blink" sketch, LED stopped working again.

### How to reproduce ?

1. download [pca10056_bootloader-0.7.0_s140_6.1.1.hex](https://github.com/adafruit/Adafruit_nRF52_Bootloader/releases/download/0.7.0/pca10056_bootloader-0.7.0_s140_6.1.1.hex) from the 0.7.0 release
2. use Programmer v3.0.3 from [nRF Connect](https://www.nordicsemi.com/Products/Development-tools/nrf-connect-for-desktop) to flash an [nRF52840-DK](https://www.nordicsemi.com/Products/Development-hardware/nrf52840-dk)
3. use Arduino 2.0.0 to flash the "Blink" example
4. observe that LED does not blink
5. use Programmer v3.0.3 to flash the ["Blinky" sample](https://docs.zephyrproject.org/2.7.0/samples/basic/blinky/README.html) from Zephyr
6. observe that LED does blink

### Screenshots

[image]

hathach, 2022-10-17:
did the programmer specify which region is out of the flash. It is probably due to the ucir register for mbr/bootloader at 0x10001014. And it should be safe to program

```
- file: 'feather_nrf52840_express_bootloader-0.7.0-8-gb55f516-dirty_s140_6.1.1.hex'
  entry: 0xF000C4E1
  data:
  - { first: 0x00000000, last: 0x00000AFF, length: 0x00000B00 }
  - { first: 0x00001000, last: 0x00025DE7, length: 0x00024DE8 }
  - { first: 0x000F4000, last: 0x000FD333, length: 0x00009334 }
  - { first: 0x000FD800, last: 0x000FD857, length: 0x00000058 }
  - { first: 0x10001014, last: 0x1000101B, length: 0x00000008 }
```

sunxkui, 2023-01-02:
after flashing the device stays in DFU-Mode without running the application.....FTHR840BOOT

DuMaM, 2023-04-18:
@hathach This is still true.
The Bootloader address starts at:
0xf4000
where the size of the compiled image is:
```  32592    1744   22558   56894    de3e _build/build-particle_xenon/particle_xenon_bootloader-0.7.0-19-g80c3a84.out
```

vivek-pai, 2023-05-15:
I'm getting this too.

Not sure if related/important nRF Programmer identifies the resulting region as an "Application" region when it's named a Bootloader region in the hex.

hathach, 2023-05-24:
it is not clear too me which part of the memory is out of region and what issue it caused

DuMaM, 2023-05-24:
@hathach add 0xf4000 + 0xde3e and you are reaching new memory region

DuMaM, 2023-05-25:
Here is a memory layout
https://infocenter.nordicsemi.com/index.jsp?topic=%2Fsdk_nrf5_v16.0.0%2Flib_bootloader.html

DuMaM, 2023-07-16:
@hathach any updates here?

hathach, 2023-07-20:
> Here is a memory layout https://infocenter.nordicsemi.com/index.jsp?topic=%2Fsdk_nrf5_v16.0.0%2Flib_bootloader.html I think current linker layout do not fit documentation.

That is not this bootloader project

> @hathach any updates here?
> 
> Bootloader should be in range of: 0x000F 8000 - 0x000F E000 (24 kB)

this bootloader is 38KB in size. There is no point to keep this open

closed 2023-07-20 by hathach